facebookresearch/habitat-lab #742: TypeError: can't assign a numpy.ndarray to a torch.ByteTensor

JiayunjieJYJ, 2021-10-01:
## Habitat-Lab and Habitat-Sim versions
habitat                 0.2.1              
habitat-sim             0.2.1

## ❓ Questions and Help
### I have questions when I run the [habitat baseline](https://github.com/facebookresearch/habitat-lab/tree/master/habitat_baselines), after I download the [test dataset ](http://dl.fbaipublicfiles.com/habitat/habitat-test-scenes.zip) and unzip it correctly and I run this command:
`python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train
`
then the problem occurs, all of the information are as follows:
`(habitat) sdu310@sdu310-Precision-5820-Tower-X-Series:~/habitat-lab$ python -u habitat_baselines/run.py --exp-config habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type train > ../jyj_personal/train
/home/sdu310/anaconda3/envs/habitat/lib/python3.6/site-packages/torch-1.9.1-py3.6-linux-x86_64.egg/torch/cuda/__init__.py:52: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 10010). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at  ../c10/cuda/CUDAFunctions.cpp:115.)
  return torch._C._cuda_getDeviceCount() > 0
2021-10-01 20:37:10,095 config: BASE_TASK_CONFIG_PATH: configs/tasks/pointnav.yaml
CHECKPOINT_FOLDER: data/new_checkpoints
CHECKPOINT_INTERVAL: -1
CMD_TRAILING_OPTS: []
ENV_NAME: NavRLEnv
EVAL:
  SPLIT: val
  USE_CKPT_CONFIG: True
EVAL_CKPT_PATH_DIR: data/new_checkpoints
FORCE_BLIND_POLICY: False
FORCE_TORCH_SINGLE_THREADED: True
LOG_FILE: train.log
LOG_INTERVAL: 10
NUM_CHECKPOINTS: 50
NUM_ENVIRONMENTS: 1
NUM_PROCESSES: -1
NUM_UPDATES: -1
ORBSLAM2:
  ANGLE_TH: 0.2617993877991494
  BETA: 100
  CAMERA_HEIGHT: 1.25
  DEPTH_DENORM: 10.0
  DIST_REACHED_TH: 0.15
  DIST_TO_STOP: 0.05
  D_OBSTACLE_MAX: 4.0
  D_OBSTACLE_MIN: 0.1
  H_OBSTACLE_MAX: 1.25
  H_OBSTACLE_MIN: 0.375
  MAP_CELL_SIZE: 0.1
  MAP_SIZE: 40
  MIN_PTS_IN_OBSTACLE: 320.0
  NEXT_WAYPOINT_TH: 0.5
  NUM_ACTIONS: 3
  PLANNER_MAX_STEPS: 500
  PREPROCESS_MAP: True
  SLAM_SETTINGS_PATH: habitat_baselines/slambased/data/mp3d3_small1k.yaml
  SLAM_VOCAB_PATH: habitat_baselines/slambased/data/ORBvoc.txt
PROFILING:
  CAPTURE_START_STEP: -1
  NUM_STEPS_TO_CAPTURE: -1
RL:
  DDPPO:
    backbone: resnet18
    distrib_backend: GLOO
    force_distributed: False
    num_recurrent_layers: 1
    pretrained: False
    pretrained_encoder: False
    pretrained_weights: data/ddppo-models/gibson-2plus-resnet50.pth
    reset_critic: True
    rnn_type: GRU
    sync_frac: 0.6
    train_encoder: True
  POLICY:
    ACTION_DIST:
      action_activation: tanh
      max_log_std: 2
      max_std: 1
[... 605 lines cut ...]
termcolor               1.1.0
terminado               0.9.1
testpath                0.4.4
threadpoolctl           2.1.0
torch                   1.9.1
tornado                 6.1
tqdm                    4.54.0
traitlets               4.3.3
typer                   0.4.0
typing-extensions       3.10.0.0
urllib3                 1.26.7
wcwidth                 0.2.5
webdataset              0.1.40
webencodings            0.5.1
Werkzeug                1.0.1
wheel                   0.37.0
widgetsnbextension      3.5.1
wrapt                   1.12.1
yacs                    0.1.8
zipp                    3.4.0
`

My Python version is Python 3.6.13.

I have been confused for a whole day, I don't know how to solve it, please help me! Thanks!!!

JiayunjieJYJ, 2021-10-06:
I dont know what dependency is wrong.

closed 2021-10-07 by erikwijmans

JiayunjieJYJ, 2021-10-07:
@erikwijmans Hi, do you mean I need to run the baseline on a GPU then I can avoid the problem?

cky783, 2022-03-08:
Hello, I am stuck in the smae issue. Do you know how to run it with gpu?