facebookresearch/habitat-lab #752: Depth sensor readings, depth sensor range, and point cloud

yhymason, 2021-10-19:
Hi there, I have a few questions regarding the depth camera in Habitat-lab:
1, I am using depth images and open3d's interfaces as following to create point cloud objects:
`
depth_img = o3d.geometry.Image(depth_as_np)
fx = camera_info_k[0][0]
fy = camera_info_k[1][1]
cx = camera_info_k[0][2]
cy = camera_info_k[1][2]
camera_intrinsic = o3d.camera.PinholeCameraIntrinsic(
     camera_info.width, camera_info.height, fx, fy, cx, cy
)
pcd = o3d.geometry.PointCloud().create_from_depth_image(
     depth=depth_img,
     intrinsic=camera_intrinsic,
     extrinsic=np.identity(4),
     depth_scale=1000.0,
     depth_trunc=1000.0,
      stride=1,
)
`
However, the point cloud generated looks too deterministic and different from what we normally expect. See the images linked below:
[image]
[image]
The first image is a point cloud generated in Habitat's test scene and the second is a Matlab point cloud example. 
As you can see, the Matlab example has a lot more randomness in terms of point coordinates. I wonder why the point cloud generated using Habitat's depth image looks different. 

2, I also wonder there's a way to limit the depth camera's FOV in its depth range and vertical axis. Since I am not interested in objects that are located on the ceiling or are far away from the robot itself.

erikwijmans, 2021-10-20:
I'm not familiar with Open3D, but `depth_scale=1000.0` seems odd. Habitat-lab outputs depth in `[0,1]` by default where a value of 1 represents something 10m away, so maybe depth_scale=10.0?

2. Not sure where that source image is from, but it looks like it's from a real camera. A simulate camera (without a noise model) will be perfect and things will be very regular.

rpartsey, 2022-09-12:
@yhymason Feel free to re-open the issue, if you still have questions.

closed 2022-09-12 by rpartsey