facebookresearch/habitat-lab #756: From camera frame to world frame using habitat.utils.geometry_utils

yhymason, 2021-10-30:
Hi there, I am trying to convert coordinates of point cloud coordinates which are initially in the depth camera's coordinate frame back to the simulator's world frame. I am not sure what's the correct way to do it here since I am not familiar with the use of quaternions in Habitat. 
I have the following 2 transformation logic:
```
            points_world[idx] = (
                quaternion_rotate_vector(depth_camera_rot, points_world[idx])
                + depth_camera_pos
            )
```
or 
```
            points_world[idx] += depth_camera_pos
            points_world[idx] = quaternion_rotate_vector(
                depth_camera_rot, points_world[idx]
            )
```
where `depth_camera_rot` is the quaternion of the depth camera, and `depth_camera_pos` is the world coordinate of the camera. 
```
        depth_camera_pos = np.asarray(depth_camera_info.position)
        depth_camera_rot = quaternion.as_quat_array(
            np.asarray(depth_camera_info.rotation)
        )
```

Can you help me confirm which logic will work?

erikwijmans, 2021-11-02:
Rotation + translation works the same with quaternions and matrices. First apply the rotation then apply the translation. 

To create the quaternion from a list, please use `habitat.utils.geometry_utils.quaternion_from_coeff` as there are different ways to store the coefficients of a quaternion and that'll make sure it's consistent.

rpartsey, 2022-09-12:
@yhymason Feel free to re-open the issue, if you still have questions.

closed 2022-09-12 by rpartsey

dhruvbatra, 2022-09-12:
@rpartsey — we get this question about Habitat coordinate system quite frequently. We should create a page or tutorial on it with visuals so that we can point people to it. I believe Eric has a draft PR on it that you might want to take over.

rpartsey, 2022-09-22:
@dhruvbatra Thanks! That's exactly what I planned to do. Opend a new Issue https://github.com/facebookresearch/habitat-lab/issues/954 and assigned to myself.