facebookresearch/habitat-lab #759: Not capable of running example files in Hbaitat-Lab

nicolaischneider, 2021-11-13:
## Not capable of running example files in Habitat-Lab
Habitat-Lab: v0.2.1
Habitat-Sim: v0.2.0
Machine: OSX 12.0.1 (macOS Monterey)

I just installed both habitat-sim and -lab on my machine as described in the instructions. I was capable of running all tests described in the habitat-sim README, however, when it came to running the example files from habitat-lab I started running into issues. Each example throws the following error after calling it:

```
(habitat_env) (env) my_mac habitat-lab % python3 examples/example.py
2021-11-13 14:59:06,563 Initializing dataset RearrangeDataset-v0
2021-11-13 14:59:06,563 Rearrange task assets are not downloaded locally, downloading and extracting now...
error: unrecognized arguments: --no-replace
usage: example.py [-h] [--uids [UIDS [UIDS ...]] | --list] [--data-path DATA_PATH] [--clean | --replace]
                  [--username USERNAME] [--password PASSWORD]

optional arguments:
  -h, --help            show this help message and exit
  --uids [UIDS [UIDS ...]]
                        Unique ID of the data to download.
  --list                List available datasource uid options and exit.
  --data-path DATA_PATH
                        Optionally provide a path to the desired root data/ directory. Default is "habitat-
                        sim/data/".
  --clean               Remove nested child directories for the datasource.
  --replace             If set, existing equivalent versions of any dataset found during download will be deleted
                        automatically. Otherwise user will be prompted before overriding existing data.
  --username USERNAME   Username to use for downloads that require authentication
  --password PASSWORD   Password to use for downloads that require authentication
```

I'm assuming that some wrong command to download the config environment is being called in the background but I wouldn't know where and why.

I tried reinstalling the packages using a new environment but I keep getting the same issue. Any help is much appreciated!

ASzot, 2021-12-14:
Hello, thanks for reaching out. Your issue should now be fixed on the most recent version. 

The `hab_suite` branch now contains most of the functionality from the paper and we are slowly merging this into `main`. We made [this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/hab_suite/examples/tutorials/colabs/Habitat2_Quickstart.ipynb) to help get started with Habitat 2.0. 

Please let me know if you have any more questions!

rpartsey, 2022-09-13:
@nicolaischneider Feel free to re-open the issue, if you still have questions.

closed 2022-09-13 by rpartsey

wastu01, 2023-03-19:
I am new to here, can we use on Mac M1 ?