facebookresearch/habitat-lab #760: Obstacles avoidance in ReplicaCAD

guhur, 2021-11-15:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1
Habitat-Sim: v0,2,1

## ❓ Questions and Help
In ReplicaCAD, I can't figure out how to find a path that avoids movable objects. 

I am loading my simulator as in `ReplicaCAD_quickstart.ipynb`, and then, I created a NavMesh as is:

```
default_nav_mesh_settings = habitat_sim.NavMeshSettings()
default_nav_mesh_settings.set_defaults()
recompute_successful = sim.recompute_navmesh(sim.pathfinder, default_nav_mesh_settings, include_static_objects=True)
```

I am finally computing the shortest path between two points with a call to `path = habitat_sim.ShortestPath()`.

The agent is going to avoid walls, but it tries to move moveable objects. Is that a setting parameter I haven't found?

Thanks,

aclegg3, 2021-11-15:
Hey @guhur,

The NavMesh in Habitat is primarily designed for static object avoidance. This is why only objects with [`MotionType::STATIC`](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.html#MotionType-STATIC) are optionally included in [`Simulator.recompute_navmesh()`](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#recompute_navmesh). If you don't intend to use object dynamics at all, you could set all objects to STATIC either in the `configs/scenes/<scene_name>.scene_instance_config.json` files, or programmatically by looping over all objects in both the [`RigidObjectManager`](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.RigidObjectManager.html) and [`ArticulatedObjectManager`](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ArticulatedObjectManager.html) and setting [`motion_type`](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletRigidObject.html#motion_type) to static. Note that you could do this once and save the result to be loaded later in future episodes ([save_nav_mesh](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#save_nav_mesh) and [load_nav_mesh](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#load_nav_mesh)).

If you intend to navigate in the presence of moving objects, you take the responsibility of managing discrepancies between the NavMesh and dynamic state of the objects onto yourself. You could recompute the navmesh during simulation, but this could result in artifacts so I don't suggest it for anything but simple, kinematic rearrangement with low frequency state change. Alternatively, you could also attach some rigid body to the agent and use discrete collision detection to filter actions which would result in contact as an additional step_filter process.

guhur, 2021-11-15:
Hello,
Thanks for your detailed answer!

closed 2021-11-15 by guhur