facebookresearch/habitat-lab #761: Can we replay a rearrangement episode?

guhur, 2021-11-16:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: hab_suite
Habitat-Sim: v0.2.1

## ❓ Questions and Help

I would like to visualize an episode of rearrangement. 

It is stored in a PDDL format, composed of base navigation and arm movements. In my understanding, the base navigation can be played by a `ShortestPathFollower`, while the arm movements can be computed from IK. 
So, I think that displaying each step of an episode is achievable, but I can't find some examples doing it.

Is that really the case? Did I miss something?

Thanks a lot for your feedback.

aclegg3, 2021-11-16:
I think this is a question for @ASzot. At a high level, if you have base positions and arm movements it should be fairly straightforward to play them back, but I don't know if we have any existing utilities for this or what format they might expect.

ASzot, 2021-11-18:
There are no existing utilities for this. However, I agree with Alex that this will be straightforward to achieve. You can use the IK actions to set the arm joints from the desired end-effector position:
* See end-effector control action here: https://github.com/facebookresearch/habitat-lab/blob/9bb7429d3618b0da4c2df50a2c3a2fdd6bd3b5b0/habitat/tasks/rearrange/actions.py#L313
* See it in use in the interactive play script: https://github.com/facebookresearch/habitat-lab/blob/hab_suite/examples/interactive_play.py

guhur, 2021-11-18:
Thanks! I will try to implement it and coming back to you.

closed 2021-11-18 by guhur

dhruvbatra, 2021-11-18:
@guhur -- sounds great. Feel free to send us a PR if you do implement this.