facebookresearch/habitat-lab #765: trainer crashed added my customised point goal sensor

yhymason, 2021-12-02:
I have been trying to train an agent that could:
Given a set of goals `S`, navigates to any goal `s_i` in `S`. 
As suggested in [this issue]( https://github.com/facebookresearch/habitat-lab/issues/735#issuecomment-928222295)
I have added the following to `nav.py`
```
@registry.register_sensor(name="MultiplePointGoalWithGPSCompassSensor")
class IntegratedMultiplePointGoalGPSAndCompassSensor(
    IntegratedPointGoalGPSAndCompassSensor
):
    r"""Sensor that integrates Multiple PointGoals observations (which are used PointGoal Navigation) and GPS+Compass.

    For the agent in simulator the forward direction is along negative-z.
    In polar coordinate format the angle returned is azimuth to the goal.
    """
    cls_uuid: str = "multiple_pointgoal_with_gps_compass"

    def get_observation(self, observations, episode, *args: Any, **kwargs: Any):
        agent_state = self._sim.get_agent_state()
        agent_position = agent_state.position
        rotation_world_agent = agent_state.rotation
        goal_positions = np.array(
            [goal.position for goal in episode.goals], dtype=np.float32
        )
        min_magnitude = float(sys.maxsize)
        closest_goal_position = None
        for goal in goal_positions:
            current_point_goal = self._compute_pointgoal(
                agent_position, rotation_world_agent, goal
            )
            current_magnitude = np.linalg.norm(current_point_goal)
            if current_magnitude < min_magnitude:
                min_magnitude = current_magnitude
                closest_goal_position = current_point_goal

        if self._goal_format == "POLAR":
            if self._dimensionality == 2:
                rho, phi = cartesian_to_polar(
                    -closest_goal_position[2], closest_goal_position[0]
                )
                return np.array([rho, -phi], dtype=np.float32)
            else:
                _, phi = cartesian_to_polar(
                    -closest_goal_position[2], closest_goal_position[0]
                )
                theta = np.arccos(
                    closest_goal_position[1] / np.linalg.norm(closest_goal_position)
                )
                rho = np.linalg.norm(closest_goal_position)
                return np.array([rho, -phi, theta], dtype=np.float32)
        else:
            if self._dimensionality == 2:
                return np.array(
                    [-closest_goal_position[2], closest_goal_position[0]],
                    dtype=np.float32,
                )
            else:
                return closest_goal_position

    def _compute_pointgoal(self, source_position, source_rotation, goal_position):
        direction_vector = goal_position - source_position
        direction_vector_agent = quaternion_rotate_vector(
            source_rotation.inverse(), direction_vector
        )
        return direction_vector_agent

```
And I reused training episodes under mp3d datasets with goals set to multiple locations instead of one single position.

Here is how my configs look like
```
# Note:  This is an example config, see habitat_baselines/config/pointnav/ppo_pointnav.yaml
# for better hyperparameters for actual training

BASE_TASK_CONFIG_PATH: "configs/training_configs/pointnav_rgbd_mp3d_training_scenes.yaml"
TRAINER_NAME: "ppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 0
VIDEO_OPTION: [ "disk", "tensorboard" ]
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
# To evaluate on all episodes, set this to -1
TEST_EPISODE_COUNT: -1
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
NUM_ENVIRONMENTS: 1
SENSORS: [ "DEPTH_SENSOR" ]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_UPDATES: -1
TOTAL_NUM_STEPS: 1e10
LOG_INTERVAL: 1
NUM_CHECKPOINTS: 50
# Force PyTorch to be single threaded as
# this improves performance considerably
FORCE_TORCH_SINGLE_THREADED: True

RL:
  PPO:
    # ppo params
    clip_param: 0.1
    ppo_epoch: 1
    num_mini_batch: 1
    value_loss_coef: 0.5
    entropy_coef: 0.01
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.5
    num_steps: 32
    hidden_size: 512
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: True
    use_linear_lr_decay: True
    reward_window_size: 50
```
and inside `pointnav_rgbd_mp3d_training_scenes.yaml` I have 
```
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
  ITERATOR_OPTIONS:
    SHUFFLE: False
    GROUP_BY_SCENE: True
    CYCLE: False
SIMULATOR:
  AGENT_0:
    SENSORS: [ 'DEPTH_SENSOR' ]
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
  DEPTH_SENSOR:
    WIDTH: 256
    HEIGHT: 256
TASK:
  TYPE: Nav-v0
  SUCCESS_DISTANCE: 0.1

  SENSORS: [ 'MultiplePointGoalWithGPSCompassSensor' ]
  MultiplePointGoalWithGPSCompassSensor:
    GOAL_FORMAT: "POLAR"
    DIMENSIONALITY: 2
    TYPE: MultiplePointGoalWithGPSCompassSensor
  GOAL_SENSOR_UUID: multiple_pointgoal_with_gps_compass

  MEASUREMENTS: [ 'DISTANCE_TO_GOAL', 'SUCCESS', 'SPL' ]
  SUCCESS:
    SUCCESS_DISTANCE: 0.1
DATASET:
  TYPE: PointNav-v1
  SPLIT: train
  DATA_PATH: data/datasets/pointnav/mp3d/v1/{split}/{split}.json.gz

VIDEO_OPTION: [ "disk" ]
```

The trainer managed to initialise successfully and trained for a few minutes. However, it failed with errors as follows

> 2021-12-02 03:58:01,024 Average window size: 50  distance_to_goal: 6.982  reward: -0.029  spl: 0.000  success: 0.000
> 2021-12-02 03:58:01,261 update: 278	fps: 146.366	
> 2021-12-02 03:58:01,261 update: 278	env-time: 21.041s	pth-time: 36.918s	frames: 8896
> 2021-12-02 03:58:01,262 Average window size: 50  distance_to_goal: 7.076  reward: -0.029  spl: 0.000  success: 0.000
> /home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo.py:173: FutureWarning: Non-finite norm encountered in torch.nn.utils.clip_grad_norm_; continuing anyway. Note that the default behavior will change in a future release to error out if a non-finite total norm is encountered. At that point, setting error_if_nonfinite=false will be required to retain the old behavior.
>   self.actor_critic.parameters(), self.max_grad_norm
> 2021-12-02 03:58:01,523 update: 279	fps: 146.262	
> 2021-12-02 03:58:01,523 update: 279	env-time: 21.158s	pth-time: 37.057s	frames: 8928
> 2021-12-02 03:58:01,523 Average window size: 50  distance_to_goal: inf  reward: nan  spl: nan  success: 0.000
> Traceback (most recent call last):
>   File "src/training/multigoal_pointnav_trainer.py", line 41, in <module>
>     trainer.train()
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/contextlib.py", line 52, in inner
>     return func(*args, **kwds)
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo_trainer.py", line 774, in train
>     self._compute_actions_and_step_envs(buffer_index)
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo_trainer.py", line 418, in _compute_actions_and_step_envs
>     step_batch["masks"],
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/policy.py", line 51, in act
>     distribution = self.action_distribution(features)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1056, in _call_impl
>     return forward_call(*input, **kwargs)
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/utils/common.py", line 75, in forward
>     return CustomFixedCategorical(logits=x)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/site-packages/torch/distributions/categorical.py", line 64, in __init__
>     super(Categorical, self).__init__(batch_shape, validate_args=validate_args)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/site-packages/torch/distributions/distribution.py", line 53, in __init__
>     raise ValueError("The parameter {} has invalid values".format(param))
> ValueError: The parameter logits has invalid values
>

erikwijmans, 2021-12-02:
```
 Average window size: 50 distance_to_goal: inf reward: nan spl: nan success: 0.000
 ```
The distance_to_goal, spl, and reward indicate that one of the training examples has no possible path for the agent. That is causing the crash.

yhymason, 2021-12-02:
@erikwijmans thanks for pointing that out, it does make sense. Is there a way I could resume training without resetting everything that's already there? If I rerun the same trainer with the same config, will it just load existing checkpoints and continue from there?

yhymason, 2021-12-02:
Also, would you please tell me what these params do exactly?
```
NUM_UPDATES: -1
TOTAL_NUM_STEPS: 1e10
LOG_INTERVAL: 1
NUM_CHECKPOINTS: 50
```

yhymason, 2021-12-03:
@erikwijmans would you please help? are there any existing examples of handling such crashes and resume training?

yhymason, 2021-12-06:
@erikwijmans @alexholdenmiller can anyone help, please?

erikwijmans, 2021-12-06:
We do have resume functionality in the latest versions. There's a `.habitat-resume-state.pth` that gets saved out to the checkpoint directory. If that exists training will restart from whatever is in that state if you just set the checkpoint location to be the same. Otherwise we don't directly support resuming as we don't save everything needed to do so in checkpoints to keep checkpoints from using a ton of disk space.

```
NUM_UPDATES: -1 # Total number of (PPO) updates. Mutually exclusive with TOTAL_NUM_STEPS
TOTAL_NUM_STEPS: 1e10 # Total number of steps of experience to train for. Mutually exclusive with NUM_UPDATEs
LOG_INTERVAL: 1 # Number of (PPO) updates between each log
NUM_CHECKPOINTS: 50 # Total number of checkpoints to save. Will be saved evenly throughout training.
```

Sorry for the slow reply, but answering GitHub issues isn't my only responsibility so while I will respond, I may be slow.

yhymason, 2021-12-07:
@erikwijmans thanks for your time and clarifications. Is there a way that in my own code I can ignore such crashes and just let the training continue? otherwise, I can only see the solution be manually monitoring the training run and resume it every time it crashes.  I am thinking about adding a `try catch` clause around the `trainer.train()` call. Do you think this would work?

yhymason, 2021-12-07:
I guess the real question here when FAIR trained the model, how did you folks handle such crashes and resumed training?

yhymason, 2021-12-07:
Also, I wonder if there's a util function in Habitat-lab that I can use to check whether there's a valid path between two positions in a given scene. Maybe I can use it to filter out unusable goal locations.

dhruvbatra, 2021-12-07:
> I guess the real question here is when FAIR trained the model, how did you folks handle such crashes and resume training?

Sorry, that is beyond the level of support we can provide. 

> Also, I wonder if there's a util function in Habitat-lab that I can use to check whether there's a valid path between two positions in a given scene. Maybe I can use it to filter out unusable goal locations.

Please go through navigation tutorials: https://aihabitat.org/tutorial/2020/

closed 2021-12-07 by dhruvbatra

yhymason, 2021-12-08:
@dhruvbatra @erikwijmans I have managed to use the sim's `_is_navigable` function and the pathfinder class to filter out all the invalid goal positions in my training data. Thanks for the help. 

However, the trainer crashed on EOF error after training for a while. The stack trace is as follows:
```
Simulator::createSceneInstance : The active scene does not contain semantic annotations. 
---
I1207 22:19:53.772099 134319 MetadataMediator.cpp:262] ::getSceneAttributesByName : Query dataset : default for SceneAttributes named : data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb yields 1 candidates.  Using data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb.
I1207 22:19:53.772109 134319 SceneDatasetAttributes.cpp:45] ::addNewSceneInstanceToDataset : Dataset : 'default' : Stage Attributes 'data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb' specified in Scene Attributes exists in dataset library.
I1207 22:19:53.772111 134319 SceneDatasetAttributes.cpp:85] ::addNewSceneInstanceToDataset : Dataset : 'default' : Lighting Layout Attributes no_lights specified in Scene Attributes exists in dataset library.
I1207 22:19:53.772117 134319 MetadataMediator.cpp:262] ::getSceneAttributesByName : Query dataset : default for SceneAttributes named : data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb yields 1 candidates.  Using data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb.
I1207 22:19:53.772122 134319 SceneDatasetAttributes.cpp:45] ::addNewSceneInstanceToDataset : Dataset : 'default' : Stage Attributes 'data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb' specified in Scene Attributes exists in dataset library.
I1207 22:19:53.772125 134319 SceneDatasetAttributes.cpp:85] ::addNewSceneInstanceToDataset : Dataset : 'default' : Lighting Layout Attributes no_lights specified in Scene Attributes exists in dataset library.
I1207 22:19:53.772260 134319 Simulator.cpp:171] ::reconfigure : createSceneInstance success == true for active scene name : data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.glb with renderer.
I1207 22:19:53.772632 134319 simulator.py:221] Loaded navmesh data/scene_datasets/mp3d/2n8kARJN3HM/2n8kARJN3HM.navmesh
2021-12-07 22:20:05,730 update: 75	fps: 95.854	
2021-12-07 22:20:05,731 update: 75	env-time: 49.058s	pth-time: 48.983s	frames: 9600
2021-12-07 22:20:05,731 Average window size: 50  distance_to_goal: 6.138  reward: 0.371  spl: 0.000  success: 0.000
I1207 22:20:16.758416 134319 PhysicsManager.cpp:50] Deconstructing PhysicsManager
I1207 22:20:16.758440 134319 SemanticScene.h:43] Deconstructing SemanticScene
Traceback (most recent call last):
  File "src/training/multigoal_pointnav_trainer.py", line 41, in <module>
    trainer.train()
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/contextlib.py", line 52, in inner
    return func(*args, **kwds)
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo_trainer.py", line 784, in train
    buffer_index
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo_trainer.py", line 458, in _collect_environment_result
    for index_env in range(env_slice.start, env_slice.stop)
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/rl/ppo/ppo_trainer.py", line 458, in <listcomp>
    for index_env in range(env_slice.start, env_slice.stop)
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/contextlib.py", line 52, in inner
    return func(*args, **kwds)
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/core/vector_env.py", line 409, in wait_step_at
    return self._connection_read_fns[index_env]()
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/core/vector_env.py", line 97, in __call__
    res = self.read_fn()
  File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/utils/pickle5_multiprocessing.py", line 68, in recv
    buf = self.recv_bytes()
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 216, in recv_bytes
    buf = self._recv_bytes(maxlength)
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
```
What does the error mean? How do I avoid it? Please help.

yhymason, 2021-12-08:
My current configs are:
```
BASE_TASK_CONFIG_PATH: "/home/lci-user/Desktop/workspace/src/ros_x_habitat/configs/training_configs/pointnav_rgbd_mp3d_training_scenes.yaml"
TRAINER_NAME: "ppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 0
VIDEO_OPTION: [ "disk", "tensorboard" ]
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
# To evaluate on all episodes, set this to -1
TEST_EPISODE_COUNT: -1
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
SENSORS: [ "DEPTH_SENSOR" ]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_UPDATES: -1
NUM_ENVIRONMENTS: 1
TOTAL_NUM_STEPS: 1e10
LOG_INTERVAL: 25
NUM_CHECKPOINTS: 100
# Force PyTorch to be single threaded as
# this improves performance considerably
FORCE_TORCH_SINGLE_THREADED: True

RL:
  PPO:
    # ppo params
    clip_param: 0.2
    ppo_epoch: 4
    value_loss_coef: 0.5
    entropy_coef: 0.01
    num_mini_batch: 1
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.5
    num_steps: 128
    hidden_size: 512
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: True
    use_linear_lr_decay: True
    reward_window_size: 50

    # Use double buffered sampling, typically helps
    # when environment time is similar or large than
    # policy inference time during rollout generation
    use_double_buffered_sampler: False
```
and
```
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
  ITERATOR_OPTIONS:
    SHUFFLE: False
    GROUP_BY_SCENE: True
    CYCLE: False
SIMULATOR:
  AGENT_0:
    SENSORS: [ 'DEPTH_SENSOR' ]
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
  DEPTH_SENSOR:
    WIDTH: 256
    HEIGHT: 256
TASK:
  TYPE: Nav-v0
  SUCCESS_DISTANCE: 0.1

  SENSORS: [ 'MultiplePointGoalWithGPSCompassSensor' ]
  MultiplePointGoalWithGPSCompassSensor:
    GOAL_FORMAT: "POLAR"
    DIMENSIONALITY: 2
    TYPE: MultiplePointGoalWithGPSCompassSensor
  GOAL_SENSOR_UUID: multiple_pointgoal_with_gps_compass

  MEASUREMENTS: [ 'DISTANCE_TO_GOAL', 'SUCCESS', 'SPL' ]
  SUCCESS:
    SUCCESS_DISTANCE: 0.1
DATASET:
  TYPE: PointNav-v1
  SPLIT: train
  DATA_PATH: data/datasets/pointnav/mp3d/v1/{split}/{split}.json.gz

VIDEO_OPTION: [ "disk" ]
```