facebookresearch/habitat-lab #766: Trainer crashed when training a fresh PPO agent on mp3d validation set

yhymason, 2021-12-07:
Hi there,

As a part of the debugging efforts towards the issues I encountered in [this post](https://github.com/facebookresearch/habitat-lab/issues/765). I have tried to set up a trainer using the mp3d validation dataset with the following configs:
ppo_pointnav.yaml:
```
BASE_TASK_CONFIG_PATH: "/home/lci-user/Desktop/workspace/src/ros_x_habitat/configs/training_configs/pointnav_rgbd_mp3d_validation_scenes.yaml"
TRAINER_NAME: "ppo"
ENV_NAME: "NavRLEnv"
SIMULATOR_GPU_ID: 0
TORCH_GPU_ID: 0
# Can be uncommented to generate videos.
VIDEO_OPTION: [ "disk", "tensorboard" ]
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
# Evaluate on all episodes
TEST_EPISODE_COUNT: -1
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
SENSORS: [ "DEPTH_SENSOR" ]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_UPDATES: -1
TOTAL_NUM_STEPS: 1e10
LOG_INTERVAL: 25
NUM_CHECKPOINTS: 100
# Force PyTorch to be single threaded as
# this improves performance considerably
FORCE_TORCH_SINGLE_THREADED: True

RL:
  PPO:
    # ppo params
    clip_param: 0.2
    ppo_epoch: 4
    value_loss_coef: 0.5
    entropy_coef: 0.01
    lr: 2.5e-4
    eps: 1e-5
    max_grad_norm: 0.5
    num_steps: 128
    hidden_size: 512
    use_gae: True
    gamma: 0.99
    tau: 0.95
    use_linear_clip_decay: True
    use_linear_lr_decay: True
    reward_window_size: 50

    # Use double buffered sampling, typically helps
    # when environment time is similar or large than
    # policy inference time during rollout generation
    use_double_buffered_sampler: False
```
pointnav_rgbd_mp3d_validation_scenes.yaml:
```
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
  ITERATOR_OPTIONS:
    SHUFFLE: False
    GROUP_BY_SCENE: True
    CYCLE: False
SIMULATOR:
  AGENT_0:
    SENSORS: [ 'DEPTH_SENSOR' ]
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
  DEPTH_SENSOR:
    WIDTH: 256
    HEIGHT: 256
TASK:
  TYPE: Nav-v0
  SUCCESS_DISTANCE: 0.1
[... 122 lines cut ...]
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 216, in recv_bytes
>     buf = self._recv_bytes(maxlength)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
>     buf = self._recv(4)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
>     raise EOFError
> EOFError
> Exception ignored in: <bound method VectorEnv.__del__ of <habitat.core.vector_env.VectorEnv object at 0x7f3d450b8cf8>>
> Traceback (most recent call last):
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/core/vector_env.py", line 588, in __del__
>     self.close()
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/core/vector_env.py", line 456, in close
>     read_fn()
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/core/vector_env.py", line 97, in __call__
>     res = self.read_fn()
>   File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat/utils/pickle5_multiprocessing.py", line 68, in recv
>     buf = self.recv_bytes()
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 216, in recv_bytes
>     buf = self._recv_bytes(maxlength)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
>     buf = self._recv(4)
>   File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/multiprocessing/connection.py", line 383, in _recv
>     raise EOFError
> EOFError: 
>

yhymason, 2021-12-07:
Does anyone have ideas about what's going on here?

yhymason, 2021-12-07:
Training with training set seems to work fine.

erikwijmans, 2021-12-08:
The default number of environments is 16, which is likely uses too much GPU or system memory causes a crash. Trying set `NUM_ENVIRONMENTS` to something like 4 in the hab baselines config.

closed 2021-12-08 by erikwijmans