facebookresearch/habitat-lab #777: Wrong paths

sparisi, 2021-12-16:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: hab_suide

Habitat-Sim: master

I am trying to run the pick tasks, but the [YAML](https://github.com/facebookresearch/habitat-lab/blob/hab_suite/configs/tasks/rearrange/pick.yaml) file points to a missing file. The file is in the `DATA_PATH` JSON
```
data/datasets/rearrange_pick/replica_cad/v0/rearrange_pick_replica_cad_v0/pick.json.gz
```
When I launch my experiments, the dataset is automatically downloaded by habitat, and the only JSON there is
```
data/datasets/rearrange_pick/replica_cad/v0/train/train_counter_L_analysis_5000_500.json.gz
```
However, this JSON has some fields that are not recognized by the `Episode` class
```
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 70, in from_json
    rearrangement_episode = RearrangeEpisode(**episode)
TypeError: __init__() got an unexpected keyword argument 'scene_config_path'
```

Not sure if I am doing something wrong, or the branch is still in dev and so the paths are broken.

ASzot, 2021-12-18:
The most recent version of Habitat Sim should download the correct files. Can you install the most recent version of the Habitat Sim `main` branch? And then run the following to force download the assets: 
```
python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets
```
I just installed from scratch and it worked fine for me.

sparisi, 2021-12-18:
I installed the latest with
`conda install -c aihabitat-nightly habitat-sim`
and ran your code. It downloaded that were missing (I guess, because if I run it a second time it asks me if I want to replace already existing file, while the first time I didn't get this prompt). However, I still get 
```
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat_baselines/utils/env_utils.py", line 28, in make_env_fn
    dataset = make_dataset(
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/datasets/registration.py", line 21, in make_dataset
    return _dataset(**kwargs)  # type: ignore
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 62, in __init__
    super().__init__(config)
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 93, in __init__
    with gzip.open(datasetfile_path, "rt") as f:
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.8/gzip.py", line 58, in open
    binary_file = GzipFile(filename, gz_mode, compresslevel)
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.8/gzip.py", line 173, in __init__
    fileobj = self.myfileobj = builtins.open(filename, mode or 'rb')
FileNotFoundError: [Errno 2] No such file or directory: 'data/datasets/rearrange_pick/replica_cad/v0/rearrange_pick_replica_cad_v0/pick.json.gz'
```

I would rather not to reinstall it from scratch, since I am currently using Habitat for other experiments and they are running fine (and in the past I had trouble setting them up, so I do not want to risk breaking my working code).

The task I am trying to run uses
```
config_file = 'habitat_baselines/config/rearrange/ddppo_pick.yaml'
base_config_file = 'configs/tasks/rearrange/pick.yaml'
```