facebookresearch/habitat-lab #779: Tutorial  "Habitat-Lab for Interaction" Colab not found

yuecao1997, 2021-12-22:
## ❓ Questions and Help
Hello, 
I tried to access the Colab for section: "Habitat-Lab for Interaction" on your tutorial page https://aihabitat.org/tutorial/2020/, 
but the Colab link https://colab.research.google.com/github/facebookresearch/habitat-lab/blob/master/examples/tutorials/colabs/Habitat_Interactive_Tasks.ipynb seems to be invaild cause the file dose not exist anymore.

Please let me how can I access this colab. 
Or any specific tutorial on how to train/eval a task would also very helpful.

dobosevych, 2022-04-01:
Check this link https://github.com/facebookresearch/habitat-lab/blob/a5acc78e28cbe2e9ca31d5442f13c48a053b6bd7/examples/tutorials/colabs/Habitat_Interactive_Tasks.ipynb
Notebook is available in commit history

rpartsey, 2022-09-13:
The file was removed in scope of PR https://github.com/facebookresearch/habitat-lab/pull/764.

rpartsey, 2023-02-28:
Notebooks/tutorials are continuously updated to be in sync with the current code. Note, that videos may be outdated because they were shot at a particular point in time. If tutorial notebook is removed it means that the functionality may be obsolete or unsupported anymore. If you are still interested in the notebook, you may always find it in the commit history.

closed 2023-02-28 by rpartsey