facebookresearch/habitat-lab #780: How to load a locally trained agent

yhymason, 2021-12-28:
Hi there, I have followed the tutorial https://aihabitat.org/tutorial/2020/ and trained a PPO agent locally. I wonder how I can specify the trained agent in my config files so that I can run evaluations on my agent.

mathfac, 2021-12-28:
Hi,
Just change run type to eval and make sure it’s loading the checkpoint you
want to evaluate in the config:

‘python -u habitat_baselines/run.py --exp-config
habitat_baselines/config/pointnav/ppo_pointnav_example.yaml --run-type eval’

yhymason, 2022-01-04:
@mathfac thanks for the reply, I tried using my checkpoint file generated by the trainer(e.g. `new_checkpoints/ckpt.99.pth). However I got the following error:
```
File "/home/lci-user/Desktop/workspace/src/habitat-lab-0.2.0/habitat_baselines/agents/ppo_agents.py", line 94, in __init__
    for k, v in ckpt["state_dict"].items()
  File "/home/lci-user/anaconda3/envs/habitat2.0/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1414, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for PointNavResNetPolicy:
	Missing key(s) in state_dict: "net.visual_encoder.running_mean_and_var._mean", "net.visual_encoder.running_mean_and_var._var", "net.visual_encoder.running_mean_and_var._count". 
```
Any idea why?

yhymason, 2022-01-06:
@erikwijmans would you please take a look as well?

rpartsey, 2022-06-21:
Hi @yhymason,
Attribute `running_mean_and_var` of `PointNavResNetPolicy.net.visual_encoder`  has attributes `_mean`, `_var`, `_count` if and only if `normalize_visual_inputs` parameter of ResNetEncoder is set to True (see [resnet_policy.py L373](https://github.com/facebookresearch/habitat-lab/blob/5ce36a6c6502fe8e86d6732ba8bab9a5db471574/habitat_baselines/rl/ddppo/policy/resnet_policy.py#L373)). And `normalize_visual_inputs` is set to True if and only if `"rgb" in observation_space.spaces` (see [ppo_agents.py L84](https://github.com/facebookresearch/habitat-lab/blob/7c8df280438f61c85d274c204612df13311021a6/habitat_baselines/agents/ppo_agents.py#L84)).

My assumption is that your checkpoint was trained with only DEPTH_SENSOR enabled, but you are trying to load it into the policy instance created with both RGB_SENSOR and DEPTH_SENSOR enabled.

More accurate answer would be possible if you could provide more details such as train/eval config files and trained checkpoint.

rpartsey, 2022-09-13:
Closing the issue because it has not had recent activity.
@yhymason Feel free to re-open the issue, if you still have questions.