facebookresearch/habitat-lab #781: How to enable two agents to independently obtain their own observations (they have their own RGB sensor and their own Depth sensor respectively)

yyf17, 2021-12-29:
How to enable two agents to independently obtain their own observations (they have their own RGB sensor and their own Depth sensor respectively)?
[image]

aclegg3, 2022-01-04:
Hey @yyf17,

This shouldn't be too tough to accomplish. You are able to create multiple agents and attach distinct sensors to each. Once the scene and agents are initialized you can manually compute and query sensor observations for any agent/sensor independently from the others. See how we do so in [this interactive viewer example](https://github.com/facebookresearch/habitat-sim/blob/11254eb05c79f0b41899ee2070a2cbfdf83c27f9/examples/viewer.py#L140). Note that doing so efficiently for training may require some refactoring in your Habitat-lab environment/task code. 

Also note that our python sensor API is undergoing some re-design in the near future. Hopefully this will make use cases like your easier, but keep an eye out for changes which may be relevant to you moving forward.

yyf17, 2022-01-05:
@aclegg3 Thank you.

closed 2022-09-13 by rpartsey