facebookresearch/habitat-lab #788: How do I get the semantic class picture not  instance segementation picture  for Replica dataset

zwy020450, 2022-01-09:
## How do I get the semantic class picture not  instance segementation picture  for Replica dataset
When i run the example.py, I can get some picture of instance segementation for the replica dataset. But  it troubles me when I want to  get the semantic class pictures which the same class objects have the same color. Can I get some help about this?

mhfy6868, 2022-01-29:
Hi, you can try the following code: 

            elif img_type == 'truth':
                img = img.numpy()

                out = np.zeros(img.shape+(3,),dtype=np.uint8)
                labels = np.unique(img)

                for label in labels:
                    mask = (img == label).nonzero()
                    color = colour_code[label]*256
                    out[mask] = color
                # # img = label2rgb(label=img, image=None,colors=colour_code, alpha=0.2, bg_label=0, kind='overlay') //about to abandon
                # plt.imshow(out)
                subdir = "semantic"
                output_dir_semantic = os.path.join(output_dir,subdir)
                imageio.imwrite('{}/sample_semantic_{}.png'.format(output_dir_semantic,"%03d"%(epoch*batch_size+i+offset)),out)

rpartsey, 2022-09-13:
Hi @zwy020450, could you please provide more details on what code are you running and what configs are you using?

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity. Feel free to re-open the issue, if you still have questions.

closed 2023-01-26 by rpartsey