facebookresearch/habitat-lab #828: How to stop Habitat printing tedious information when frequently instantiating habitat.VectorEnv objects?

ZJULiHongxin, 2022-03-10:
## ❓ Questions and Help

Whenever I am running my RL training code in which habitat.VectorEnv has to be frequently instantiated to generate envs, I get a long list of redundant information about the simulation env. These tedious printouts have totally occupied the whole screen of the terminal, hindering me from checking the progress of training rewards.

A fraction of the env information that I don't know how to turn off is shown below:

2022-03-10 20:56:14,570 initializing sim Sim-0
ANGLE: 0
HEIGHT: 64
HFOV: 30
ORIENTATION: [0, 2.6179938779914944, 0]
POSITION: [0, 0.88, 0]
SENSOR_SUBTYPE: PINHOLE
TYPE: PanoramicPartRGBSensor
WIDTH: 21
ANGLE: 0
HEIGHT: 64
....
ORIENTATION: [0, -3.141592653589793, 0]
POSITION: [0, 0.88, 0]
SENSOR_SUBTYPE: PINHOLE
TYPE: PanoramicPartDepthSensor
WIDTH: 21
WARNING: Logging before InitGoogleLogging() is written to STDERR

Finally, I managed to locate where these printouts are generated. It is "return _sim(**kwargs)" in line 20 of habitat/sims/registration.py
Unfortunately, I have no idea what to do next to turn it off.

Could anyone help me stop habitat printing the info when generating simulation envs, so I am able to get a cleaner training log on the terminal?

Thanks for any solution!

ZJULiHongxin, 2022-03-26:
I found a solution. 
Put this line in the code and Habitat-sim will stop printing redundant information.
`os.environ['HABITAT_SIM_LOG'] = "quiet"`

closed 2022-03-26 by ZJULiHongxin

GuoPingPan, 2022-12-24:
@ZJULiHongxin 
Hi I want to know how to use the function of my own constuction env -- `Exploration_Env`
When I put it in to VectorEnv for multi-process, and want to use the function of  `Exploration_Env`, like `get_short_term_goal`

```
    envs = VectorEnv(
        make_env_fn=make_env_fn,
        env_fn_args=tuple(
            tuple(
                zip(args_list, env_configs, baseline_configs,
                    range(args.num_processes))
            )
        ),
    )
```

```
def make_env_fn(args, config_env, config_baseline, rank):
    dataset = PointNavDatasetV1(config_env.DATASET)
    config_env.defrost()
    config_env.SIMULATOR.SCENE = dataset.episodes[0].scene_id
    print("Loading {}".format(config_env.SIMULATOR.SCENE))
    config_env.freeze()

    env = Exploration_Env(args=args, rank=rank,
                          config_env=config_env, config_baseline=config_baseline, dataset=dataset
                          )

    env.seed(rank)
    return env
```
I implement the function `get_short_term_goal` in `Exploration_Env`, but got the error as below.
AttributeError: 'VectorEnv' object has no attribute 'get_short_term_goal'

> Can you offer me some tutorial for using VectorEnv?
> Thanks a lot.