facebookresearch/habitat-lab #835: Problem with loading semantic from MP3D dataset

alstar8, 2022-03-23:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master
Habitat-Sim: master

## 🐛 Bug

When I load the MP3D dataset, the semantic seems rotated and get a view outside from building

## Steps to Reproduce

Notebook to reproduce this issue
[habitat_example.zip](https://github.com/facebookresearch/habitat-lab/files/8335824/habitat_example.zip)

[image]

## Additional context

With stable version of Habitat I have no such an issue

dhruvbatra, 2022-03-23:
CC: @jturner65. Looks like a config issue.

alstar8, 2022-03-31:
@dhruvbatra @jturner65 Still got that problem, any advice what to change in config to fix that?

Jthon-lab, 2022-07-13:
I've got the same problem, do you know how to change the config? Thank you @alstar8.

aclegg3, 2022-07-13:
https://github.com/facebookresearch/habitat-sim/issues/1808

closed 2022-09-13 by rpartsey

MlLearnerAkash, 2026-02-15:
I also have the same issue.
Did anybody solve that?