facebookresearch/habitat-lab #849: New environments make Conv2d very slow

sparisi, 2022-04-09:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: main branch, up-to-date
Habitat-Sim: 0.2.1

## 🐛 Bug
If I create more than one Habitat env and use `Conv2d` layers for other things, the code because extremely slow. More details below.

## Steps to Reproduce

```
import time
import pathlib
import os
import torch
from torch import nn

os.environ['MAGNUM_LOG'] = 'quiet'
os.environ['GLOG_minloglevel'] = '2'
os.environ['HABITAT_SIM_LOG'] = 'quiet'

from habitat_baselines.config.default import get_config
from habitat_baselines.common.environments import get_env_class
from habitat_baselines.utils.env_utils import make_env_fn as make_env_habitat

abs_root = pathlib.Path(__file__).parent.resolve()
config_file = os.path.join(abs_root, 'habitat_config', 'nav_task.yaml')

def test(scenes, model):
    for i, scene in enumerate(scenes):
        config = get_config(config_paths=config_file,
            opts=['BASE_TASK_CONFIG_PATH', config_file])

        config.defrost()
        config.TASK_CONFIG.DATASET.DATA_PATH = \
            os.path.join(abs_root, config.TASK_CONFIG.DATASET.DATA_PATH)
        config.TASK_CONFIG.DATASET.SCENES_DIR = \
            os.path.join(abs_root, config.TASK_CONFIG.DATASET.SCENES_DIR)
        config.TASK_CONFIG.DATASET.SCENES_DIR += scene
        config.freeze()

        env_class = get_env_class(config.ENV_NAME)
        env = make_env_habitat(env_class=env_class, config=config)

        t = time.time()
        for episode in range(20):
            env.reset()
            for step in range(500):
                env.step(**{'action': 1})
                model(torch.rand(1, 3, 64, 64))
        print(i, time.time() - t)
        env.close()

if __name__ == '__main__':
    scenes = ['apartment_0', 'room_0', 'hotel_0']
    model = nn.Sequential(
        nn.Conv2d(in_channels=3, out_channels=32, kernel_size=(3, 3), stride=2, padding=1),
        nn.ELU(),
        nn.Conv2d(in_channels=32, out_channels=32, kernel_size=(3, 3), stride=2, padding=1),
        nn.ELU(),
        nn.Conv2d(in_channels=32, out_channels=32, kernel_size=(3, 3), stride=2, padding=1),
        nn.ELU(),
        nn.Conv2d(in_channels=32, out_channels=32, kernel_size=(3, 3), stride=2, padding=1),
        nn.ELU(),
        nn.Conv2d(in_channels=32, out_channels=32, kernel_size=(3, 3), stride=2, padding=1),
        nn.ELU(),
    )
    test(scenes, model)
```

I am using Replica scenes.  
`nav_task.yaml` is attached as `txt`.

## Expected behavior
All loops over the scenes should take roughly the same time, but 2nd and 3rd loop take much longer, regardless of the scenes order.

## Additional context
After the second environment is created everything becomes much slower, usually x10.
* The problem disappear if there is no `Conv2d` layers. Removing `ELU` does not help. Reducing the number of `Conv2d` layers does not help. Using only `nn.Linear` layers doesn't cause any slowdown. 
* The problem does not appear with other environments (e.g., `env = gym.make('Pendulum-v1')`).
* The problem is due to the creation of a new Habitat env. If I keep the first env going for more episode the time of the first loop scales linearly (on my machine 10 episodes of 500 steps take 10s, 20 episodes 20 seconds, and so on...). It's always the second environment to be much slower.
* The env order does not matter, i.e., changing the order of `scenes` does not matter.

I have no idea why this happens. How could the creation of a Habitat env 'mess' the performance of `Conv2d`? Habitat is installed for GPU support and automatically runs of `cuda:0`, but the network is on CPU.
(Btw, can I manually set Habitat to run on CPU?) 
[nav_task.txt](https://github.com/facebookresearch/habitat-lab/files/8457337/nav_task.txt)

erikwijmans, 2022-04-10:
Huh, this is really quite odd. My guess is that you are bumping up against either GPU or CPU RAM limits and for whatever reason the second env causes thrashing. Does `htop` or `nvidia-smi` show that you are near/over the limits on either of those?

> (Btw, can I manually set Habitat to run on CPU?)

No. While the renderer can theoretically run with a CPU software rasterizer like LLVMpipe, it's useless slow so we haven't put any engineering effort into being able to run that.

sparisi, 2022-04-10:
Just to add few more things.  

I checked what takes time within the loop, if it is the model forward or the env step.  
It is the former and these are the times (total time, avg env step time, avg model fwd time):  
scene 0: 5.394575834274292 0.001211775302886963 0.0009212926864624023
scene 1: 60.74009418487549 0.0009357562065124512 0.023349629592895507
scene 2: 59.83483290672302 0.0010255617141723633 0.022897001934051513

I also checked moving the model on GPU and the problem disappears:
scene 0: 3.5148046016693115 0.000768497085571289 0.0006268096923828125
scene 1: 2.945328950881958 0.0005623345375061035 0.0006103537559509277
scene 2: 3.0039291381835938 0.0005787138938903809 0.0006178741455078125

Regarding `nvidia-smi` and `htop`, they both seem ok (checked with model on CPU).
The former looks like this for the whole execution

```
|===============================+======================+======================|
|   0  Quadro GP100        On   | 00000000:AF:00.0 Off |                    0 |
| 26%   37C    P0    31W / 235W |     52MiB / 16278MiB |      1%      Default |
|===============================+======================+======================|
```

The latter shows increasing MEM, but at a very slow rate (never goes above 6.2GB). I have read in another issue here that this is normal. Number of procs is also stable.

Thanks for your help!

erikwijmans, 2022-04-10:
There is definitely some weird interaction between PyTorch's CPU convolution and Habitat going on here, but I unfortunately have no idea where to even begin with debugging that. If others do, feel free to comment here.