facebookresearch/habitat-lab #855: The images of the custom rgb and Depth sensor are in sync, how to solve it?

yyf17, 2022-04-24:
## 🚀 
The images of the custom rgb and Depth sensor are in sync, how to solve it?

[image]

erikwijmans, 2022-04-24:
We'll need more details to be helpful here. Can you post a code snippet of how you are creating the cameras and how you are moving them?

closed 2022-04-25 by dhruvbatra

yyf17, 2022-04-25:
@erikwijmans 
The definition of the sensor is:
# RGBSensorForAgent0
```
@registry.register_sensor(name="RGBSensorForAgent0")
#class RGBSensorForAgent0(Sensor):   #, HabitatSimRGBSensor
class RGBSensorForAgent0(RGBSensor): 
    cls_uuid: str = "rgb_for_Agent0"

    def __init__(
        self, *args: Any, sim: Simulator, config: Config, **kwargs: Any
    ):
        self._sim = sim
        sensors = self._sim.sensor_suite.sensors
        rgb_sensor_uuids = [
            uuid
            for uuid, sensor in sensors.items()
            if isinstance(sensor, RGBSensor)
        ]
        #print("rgb_sensor_uuids:soundspaces/tasks/RIR_Sensors.py",rgb_sensor_uuids)
        #print("sensors:",sensors)
        if len(rgb_sensor_uuids) != 1:
            raise ValueError(
                f"RGBSensorForAgent0 requires one RGB sensor, {len(rgb_sensor_uuids)} detected"
            )

        (self._rgb_sensor_uuid,) = rgb_sensor_uuids
        
        #print("uuid for RGBSensorForAgent0 is: ",self._rgb_sensor_uuid)
        
        super().__init__(config=config)

    def _get_uuid(self, *args: Any, **kwargs: Any) -> str:
        return self.cls_uuid

    def _get_sensor_type(self, *args: Any, **kwargs: Any):
        return SensorTypes.PATH

    def _get_observation_space(self, *args: Any, **kwargs: Any):
        return self._sim.sensor_suite.observation_spaces.spaces[
            self._rgb_sensor_uuid
        ]

    def get_observation(
        self,
        *args: Any,
        observations,
        episode: NavigationEpisode,
        **kwargs: Any,
    ):
        agent_id = 0
        current_state = self._sim.get_agent_state(agent_id)

        print(f"RGBSensorForAgent0  |agent_id:{agent_id},position:{current_state.position},rotation:{current_state.rotation}")

        cur_observation = self._sim.get_observations_at(
            position=current_state.position.tolist(), rotation=current_state.rotation.tolist(), \
                keep_agent_at_new_pose=False, agent_id = agent_id
        )

        return cur_observation[self._rgb_sensor_uuid]
```
# 
```
@registry.register_sensor(name="DepthSensorForAgent0")
#class DepthSensorForAgent0(Sensor):  #HabitatSimDepthSensor
class DepthSensorForAgent0(DepthSensor):
    cls_uuid: str = "depth_for_Agent0"

    def __init__(
        self, *args: Any, sim: Simulator, config: Config, **kwargs: Any
    ):
        self._sim = sim
        sensors = self._sim.sensor_suite.sensors

        depth_sensor_uuids = [
            uuid
            for uuid, sensor in sensors.items()
            if isinstance(sensor, DepthSensor)
        ]
        if len(depth_sensor_uuids) != 1:
            raise ValueError(
                f"DepthSensorForAgent0 requires one Depth sensor, {len(depth_sensor_uuids)} detected"
            )

        (self._depth_sensor_uuid,) = depth_sensor_uuids
        self._current_episode_id: Optional[str] = None
        
        #print("uuid for DepthSensorForAgent0 is: ", self._depth_sensor_uuid)  #  depth

        super().__init__(config=config)

    def _get_uuid(self, *args: Any, **kwargs: Any) -> str:
        return self.cls_uuid

    def _get_sensor_type(self, *args: Any, **kwargs: Any):
        return SensorTypes.PATH

    def _get_observation_space(self, *args: Any, **kwargs: Any):
        return self._sim.sensor_suite.observation_spaces.spaces[
            self._depth_sensor_uuid
        ]

    def get_observation(
        self,
        *args: Any,
        observations,
        episode: NavigationEpisode,
        **kwargs: Any,
    ):
        agent_id = 0
        current_state = self._sim.get_agent_state(agent_id)

        print(f"DepthSensorForAgent0|agent_id:{agent_id},position:{current_state.position},rotation:{current_state.rotation}")

        cur_observation = self._sim.get_observations_at(
            position=current_state.position.tolist(), rotation=current_state.rotation.tolist(), \
                keep_agent_at_new_pose=False, agent_id = agent_id
        )

        return cur_observation[self._depth_sensor_uuid]

```

erikwijmans, 2022-04-25:
```
self._sim.get_observations_at(
            position=current_state.position.tolist(), rotation=current_state.rotation.tolist(), \
                keep_agent_at_new_pose=False, agent_id = agent_id
        )
```

Two things: 

1. current_state.rotation.tolist() isn't correct, please use `habitat.utils.geometry_utils.quaternion_to_list`
2. We don't have an `agent_id` arg in `get_observations_at` so presumably you are using a modified version of the code base. Likely those modifications are the source of the bug.

yyf17, 2022-04-25:
@erikwijmans 
Thank you.
I have modified Simulator
```
    def step(self, action, only_allowed=True):
        """
        All angle calculations in this function is w.r.t habitat coordinate frame, on X-Z plane
        where +Y is upward, -Z is forward and +X is rightward.
        Angle 0 corresponds to +X, angle 90 corresponds to +y and 290 corresponds to 270.

        :param action: action to be taken
        :param only_allowed: if true, then can't step anywhere except allowed locations
        :return:
        Dict of observations
        """
        assert self._is_episode_active, (
            "episode is not active, environment not RESET or "
            "STOP action called previously"
        )

        self._previous_step_collided = False
        # STOP: 0, FORWARD: 1, LEFT: 2, RIGHT: 2
        if action == HabitatSimActions.STOP:
            self._is_episode_active = False
        else:
            pass

        position_list_Agent0 = list(self.graph.nodes[self._position_index_Agent0]['point'])
        position_list_Agent1 = list(self.graph.nodes[self._position_index_Agent1]['point'])

        rotation_quat_Agent0 = quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent0), np.array([0, 1, 0]))
        rotation_quat_Agent1 = quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent1), np.array([0, 1, 0]))

        if self.config.USE_RENDERED_OBSERVATIONS:
            self._sim.set_agent_state(position_list_Agent0, rotation_quat_Agent0, agent_id = 0)
            self._sim.set_agent_state(position_list_Agent1, rotation_quat_Agent1, agent_id = 1)
        else:
            self.set_agent_state(position_list_Agent0, rotation_quat_Agent0, agent_id = 0)
            self.set_agent_state(position_list_Agent1, rotation_quat_Agent1, agent_id = 1)

            #

        # print("0---------------------")
        # print("list(self.graph.nodes[self._position_index_Agent0]['point']):",list(self.graph.nodes[self._position_index_Agent0]['point']))
        # print("quat_to_coeffs(quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent0), np.array([0, 1, 0]))).tolist()",quat_to_coeffs(quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent0), np.array([0, 1, 0]))).tolist())
        # new_state = self.get_agent_state(0)
        # print("0:position:",new_state.position)
        # print("0:rotation:",new_state.rotation)
        # print("1---------------------")
        # print("list(self.graph.nodes[self._position_index_Agent1]['point']):",list(self.graph.nodes[self._position_index_Agent1]['point']))
        # print("quat_to_coeffs(quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent1), np.array([0, 1, 0]))).tolist()",quat_to_coeffs(quat_from_angle_axis(np.deg2rad(self._rotation_angle_Agent1), np.array([0, 1, 0]))).tolist())
        # new_state = self.get_agent_state(1)
        # print("0:position:",new_state.position)
        # print("0:rotation:",new_state.rotation)

        
        self._episode_step_count += 1

        sim_obs = self._get_sim_observation()
        if self.config.USE_RENDERED_OBSERVATIONS:
            self._sim.set_sensor_observations(sim_obs)
        self._prev_sim_obs = sim_obs
        observations = self._sensor_suite.get_observations(sim_obs)
        
        if self.config.CONTINUOUS_VIEW_CHANGE:
            observations['intermediate'] = []
            for i in range(len(self._intermediate_Agent0)):
                obs = {
                    'rgb_for_Agent0':self._intermediate_Agent0[i]['rgb_for_Agent0'],
                    'rgb_for_Agent1':self._intermediate_Agent1[i]['rgb_for_Agent1'],
[... 191 lines cut ...]
    def set_agent_state(
        self,
        position: List[float],
        rotation: List[float],
        agent_id: int = 0,
        reset_sensors: bool = True,
    ):

        if not self.config.USE_RENDERED_OBSERVATIONS:
            agent = self._sim.get_agent(agent_id)
            new_state = self.get_agent_state(agent_id)
            new_state.position = position
            new_state.rotation = rotation
            # NB: The agent state also contains the sensor states in _absolute_
            # coordinates. In order to set the agent's body to a specific
            # location and have the sensors follow, we must not provide any
            # state for the sensors. This will cause them to follow the agent's
            # body
            new_state.sensor_states = {}
            agent.set_state(new_state, reset_sensors)
            return True
            
        else:
            return self._sim.set_agent_state(position, rotation, agent_id)
```

erikwijmans, 2022-04-25:
If you can make a minimal reproduction of the issue I'm happy to look into it but this is too large of change-set for me to realistically look into.

yyf17, 2022-04-26:
@erikwijmans I will try to make a minimal reproduction of the issue.
Thank you.