facebookresearch/habitat-lab #856: Could you refer me to something that allows generating expert (maybe suboptimal) demonstrations for navigation tasks?

manjrekarom, 2022-04-25:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1
Habitat-Sim: v0.2.1

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ 
Yes

## ❓ Questions and Help
Greetings, 
I am trying to implement a [paper](https://arxiv.org/abs/2204.02372) and I want to bootstrap with a policy trained with behavior cloning. Is there a way to collect an offline dataset of demonstrations automatically from habitat. I think I would need to extend the idea shown in one of the [tutorials](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb)

These I think are the steps (could be more than one but I want to make it easier for me):
```
# repeat k times for k trajectories
env.reset()  # randomly sample an episode
sim = env.habitat_env.sim
# follow the simulator tutorial using NavMesh to find shortest path to goal
# find actions and observations that take you to that goal
```

I want to know if my approach is okay and if there are APIs (if so could you point me to them) to get the goal point (unlike random in the tutorial) and find actions to the goal (getting observations is covered in the tutorial).

erikwijmans, 2022-04-25:
This isn't something we have an example of, but you can take a look at the dagger trainer in https://github.com/jacobkrantz/VLN-CE as a starting point.

closed 2022-04-25 by erikwijmans

manjrekarom, 2022-04-28:
@erikwijmans thanks for your reply. Could you briefly describe what dagger trainer is doing as the dagger that I am aware of needs human in the loop?

Do you think it would be easier for me to use a pre-trained model to do that instead? If so are there any RGB models for PointNav? I saw some trained on Gibson but I am working with hm3d.