facebookresearch/habitat-lab #857: No depth key in observations dict even when it's set in config

manjrekarom, 2022-04-25:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.1

Habitat-Sim: v0.2.1

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes

## ❓ Questions and Help
I trained ppo with the following code and config (ignore the val dataset used for training part).
```
config = get_baselines_config(
    "./habitat_baselines/config/pointnav/ppo_pointnav.yaml"
)

config.defrost()
config.NUM_UPDATES = -1
config.NUM_CHECKPOINTS = 5
config.TASK_CONFIG.DATASET.DATA_PATH = './data/datasets/pointnav/hm3d/v1/{split}/{split}.json.gz'
config.TASK_CONFIG.DATASET.SPLIT = 'val'
config.freeze()

random.seed(config.TASK_CONFIG.SEED)
np.random.seed(config.TASK_CONFIG.SEED)

if __name__ == "__main__":
    trainer_init = baseline_registry.get_trainer(config.TRAINER_NAME)
    trainer = trainer_init(config)
    trainer.train()
```

Now I am trying to run the agent using `habitat-lab/habitat_baselines/agents/ppo_agents.py` like this:
`python habitat_baselines/agents/ppo_agents.py  --model-path ./data/new_checkpoints/ckpt.4.pth --input-type depth --task-config ./habitat_baselines/config/pointnav/ppo_pointnav.yaml`

But I get the following error:
```
2022-04-25 03:27:39,086 Initializing task Nav-v0
Traceback (most recent call last):
  File "habitat_baselines/agents/ppo_agents.py", line 172, in <module>
    main()
  File "habitat_baselines/agents/ppo_agents.py", line 165, in main
    metrics = benchmark.evaluate(agent)
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat/core/benchmark.py", line 169, in evaluate
    return self.local_evaluate(agent, num_episodes)
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat/core/benchmark.py", line 139, in local_evaluate
    action = agent.act(observations)
  File "habitat_baselines/agents/ppo_agents.py", line 130, in act
    ) = self.actor_critic.act(
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat_baselines/rl/ppo/policy.py", line 68, in act
    features, rnn_hidden_states = self.net(
  File "/home/omanjrekar/miniconda3/envs/rl/lib/python3.8/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat_baselines/rl/ddppo/policy/resnet_policy.py", line 370, in forward
    visual_feats = self.visual_encoder(observations)
  File "/home/omanjrekar/miniconda3/envs/rl/lib/python3.8/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat_baselines/rl/ddppo/policy/resnet_policy.py", line 186, in forward
    depth_observations = observations["depth"]
  File "/home/omanjrekar/MBRL-VLN/habitat-lab/habitat_baselines/common/tensor_dict.py", line 67, in __getitem__
    return super().__getitem__(index)
KeyError: 'depth'
```

I don't quite understand why I don't have the 'depth' key in observations. This result is similar for the following code where I instantiate `NavRLEnv` and get observations (where depth key doesn't exist).

```
import os
import cv2
import random

import habitat
from habitat_baselines.common.baseline_registry import baseline_registry
from habitat_baselines.config.default import get_config 

habitat_lab_base_dir = '/home/abcd/MBRL-VLN/habitat-lab/'
os.chdir(habitat_lab_base_dir)

# specify config
config = get_config(os.path.join(habitat_lab_base_dir, 'configs/tasks/pointnav.yaml'))

config.defrost()
config.TASK_CONFIG.DATASET.SPLIT = 'train_10_percent'
config.TASK_CONFIG.DATASET.DATA_PATH = './data/datasets/pointnav/hm3d/v1/{split}/{split}.json.gz'
config.SIMULATOR.AGENT_0.SENSORS = ['RGB_SENSOR', 'DEPTH_SENSOR']
config.freeze()

NavRLEnv = baseline_registry.get_env('NavRLEnv')
env = NavRLEnv(config)

# explore env
print('Action space:', env.action_space.spaces)
print('Obs Space:', env.observation_space.spaces)

obs = env.reset()
done = False
i = 0
while not done:
    i += 1
    action = {'action': random.choice(['MOVE_FORWARD', 'TURN_LEFT', 'TURN_RIGHT'])}
    # print(action)
    next_obs, reward, done, info = env.step(action=action)

print(next_obs.keys())

env.close()
```

The output:
```
# config truncated to show important parts
SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR']
SIMULATOR:
  AGENT_0:
    SENSORS: ['RGB_SENSOR', 'DEPTH_SENSOR']
  DEPTH_SENSOR:
    HEIGHT: 256
    WIDTH: 256
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
  RGB_SENSOR:
    HEIGHT: 256
    WIDTH: 256
SIMULATOR_GPU_ID: 0
TASK:
  GOAL_SENSOR_UUID: pointgoal_with_gps_compass
  MEASUREMENTS: ['DISTANCE_TO_GOAL', 'SUCCESS', 'SPL']
  POINTGOAL_WITH_GPS_COMPASS_SENSOR:
    DIMENSIONALITY: 2
    GOAL_FORMAT: POLAR
  SENSORS: ['POINTGOAL_WITH_GPS_COMPASS_SENSOR']
  SUCCESS:
    SUCCESS_DISTANCE: 0.2
  SUCCESS_DISTANCE: 0.2
  TYPE: Nav-v0

# keys in next_obs
dict_keys(['rgb', 'pointgoal_with_gps_compass'])
```

I feel like I am missing something.

Skylion007, 2022-05-12:
@manjrekarom Habitat Baselines tasks currently overrides sensors: https://github.com/facebookresearch/habitat-lab/blob/323f78684de0d348fc57f0cd3be4ae8a5c0e3435/habitat_baselines/config/pointnav/ddppo_pointnav.yaml#L13 . It's something we are definitely working on fixing in our tech debt.

manjrekarom, 2022-05-12:
Thanks for your reply. So shouldn't I have only "depth" and not "rgb" in observations, if the above config overrides it?

rpartsey, 2022-09-13:
Hi @manjrekarom, yes you are right if you used this config [habitat-lab/habitat_baselines/config/pointnav/ddppo_pointnav.yaml](https://github.com/facebookresearch/habitat-lab/blob/323f78684de0d348fc57f0cd3be4ae8a5c0e3435/habitat_baselines/config/pointnav/ddppo_pointnav.yaml#L13) then you would have only "depth" key  in observations.

1. The problem with your code is that you are using this config file: https://github.com/facebookresearch/habitat-lab/blob/00ee28f8339a0b20966f12456728676256ebc246/configs/tasks/pointnav.yaml#L5
That's why you have "rgb" key in observations but don't have "depth" key.

2. Regarding your example code
> from habitat_baselines.config.default import get_config 
> config = get_config(os.path.join(habitat_lab_base_dir, 'configs/tasks/pointnav.yaml'))

This code creates the baseline config, passing it to the environment constructor `env = NavRLEnv(config)` is not valid

You should do this `Env(config.TASK_CONFIG)`.

rpartsey, 2023-01-26:
Closing the issue because it has not had recent activity. Feel free to re-open the issue, if you still have questions.

closed 2023-01-26 by rpartsey