facebookresearch/habitat-lab #871: Pre-trained models for Habitat 2.0 Skills

arjung128, 2022-05-20:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hi, would it be possible to make pre-trained models for the Habitat 2.0 skills available for use? I see some in [here](https://drive.google.com/drive/folders/1ILCSiJVKloTRnKdpB5Qdn7FEo6ge3PQM), but not the ones I'm looking for (e.g. Open Drawer / Open Fridge)...

aclegg3, 2022-05-23:
@ASzot are we planning to release any of these in the near future?

RobertMcCarthy97, 2023-01-03:
Just following up on this - Access to the models from the Habitat 2.0 paper (ideally in a form compatible with the latest stable releases) would be very useful for me also!

ASzot, 2023-01-05:
Yes, we have released a set of pre-trained models. Instructions for downloading the pre-trained models and using them can be found in this [README for the Habitat Rearrangement Challenge](https://github.com/facebookresearch/habitat-challenge/tree/rearrangement-challenge-2022#hierarchical-rl-starter-code).

RobertMcCarthy97, 2023-01-09:
Great, thank you - these worked for me!