facebookresearch/habitat-lab #893: TASK.SUCCESS.SUCCESS_DISTANCE v.s. TASK.SUCCESS_DISTANCE

sagadre, 2022-06-28:
## ❓ Questions and Help

Based on the [habitat challenge 2022 object nav page](https://aihabitat.org/challenge/2022/), seems like the success distance should be 1.0m.

Looking at the default config I see the following:

```
TASK:
    SUCCESS:
        SUCCESS_DISTANCE: 0.1
        TYPE: Success
    SUCCESS_DISTANCE: 1.0
```

Just want to make sure that `TASK.SUCCESS_DISTANCE` is being used and not `TASK.SUCCESS.SUCCESS_DISTANCE` especially in light of this [PR](https://github.com/facebookresearch/habitat-lab/pull/784) suggesting `TASK.SUCCESS_DISTANCE` is supposed to be deprecated.

Tried digging around the code, but am not familiar with all the abstractions, so I decided to ask😄 

Thanks!

sagadre, 2022-06-28:
After playing around a bit, it does seem like `TASK.SUCCESS.SUCCESS_DISTANCE` is being used.
Does this mean that the success distance is 10.0cm and not 1.0m?

dhruvbatra, 2022-06-29:
@ykarmesh @srama2512 @devendrachaplot

devendrachaplot, 2022-06-29:
The success distance is 1m. I believe 1m is used to find a set valid points around the object, 0.1m is the success threshold from any of these valid points. I believe @Skylion007 implemented this, is my understanding correct?

Skylion007, 2022-06-29:
I think TASK.SUCCESS_DISTANCE is deprecated, but may be used for ObjectNav. I believe @devendrachaplot understanding is correct. It really shouldn't be used for this purpose, we do need to refactor so that we have a `TASK.OBJ_NAV.SUCCESS_DISTANCE` for better documentation and composability. This is definitely part of the config we should redesign as we continue the HabLab refactor that @vincentpierre is discussing.

closed 2022-06-30 by sagadre