facebookresearch/habitat-lab #896: Where can I find the robot URDF file？Such as "./data/robots/hab_fetch/robots/hab_fetch.urdf"

xuexidi, 2022-07-06:
## ❓ Questions and Help
Hello!
I would like to train an HAB agent under replica_cad dataset (nav-pick-nav-place). 
But I can't found files described in **<habitat-lab/configs/tasks/rearrange/nav_pick_nav_place.yaml>** such as: "_DATA_PATH: data/datasets/replica_cad/rearrange/v1/{split}/all_receptacles_10k_1k.json.gz_", "_ROBOT_URDF: ./data/robots/hab_fetch/robots/hab_fetch.urdf_" and so on.

Where can I download these files？Looking forward to your help, thank you!

aclegg3, 2022-07-06:
Hey @xuexidi,

You can get these assets from our S3 hosting. The easiest way is to use the datasets_download.py utility. Specifically, the [rearrange_task_assets](https://github.com/facebookresearch/habitat-sim/blob/127d26b4a226493c9bbefdc7353e5bc5fb44963c/src_python/habitat_sim/utils/datasets_download.py#L273) in this case.

With habitat-sim module installed in your conda env:
`python -m habitat_sim.utils.datasets_download --uids rearrange_task_assets --data-path /path/to/data/`

closed 2022-07-06 by aclegg3

xuexidi, 2022-07-08:
> Hey @xuexidi,
>
> You can get these assets from our S3 hosting. The easiest way is to use the datasets_download.py utility. Specifically, the [...]

@aclegg3 
OK, I'll try it！Thanks a lot ! ! !