facebookresearch/habitat-lab #908: [Help!]ConnectionResetError: [Errno 104] Connection reset by peer

xuexidi, 2022-07-20:
I train the Habitat Object nav use DDPPO baselines normally in my Ubuntu 18.04 server.
When I back up the conda virtual environment and training code to other Ubuntu machines (Ubuntu 18.04 server, too) for training, an error appears in the title. What is the possible reason?

`---
I0720 20:43:58.673173 30334 simulator.py:213] Loaded navmesh data/scene_datasets/mp3d/VFuaQ6m2Qom/VFuaQ6m2Qom.navmesh
I0720 20:43:58.673994 30334 simulator.py:225] Recomputing navmesh for agent's height 0.88 and radius 0.18.
I0720 20:43:58.686116 30334 PathFinder.cpp:382] Building navmesh with 704x949 cells
I0720 20:43:59.103587 30334 PathFinder.cpp:650] Created navmesh with 2584 vertices 1279 polygons
I0720 20:43:59.103703 30334 Simulator.cpp:710] reconstruct navmesh successful
2022-07-20 20:43:59,111 Initializing task ObjectNav-v1
Traceback (most recent call last):
  File "habitat_baselines/run.py", line 80, in <module>
    main()
  File "habitat_baselines/run.py", line 40, in main
    run_exp(**vars(args))
  File "habitat_baselines/run.py", line 76, in run_exp
    execute_exp(config, run_type)
  File "habitat_baselines/run.py", line 59, in execute_exp
    trainer.train()
  File "/home/xuexidi/anaconda3/envs/embclip-habitat/lib/python3.6/contextlib.py", line 52, in inner
    return func(*args, **kwds)
  File "/home/xuexidi/xue/embclip-habitat/habitat_baselines/rl/ppo/ppo_trainer.py", line 724, in train
    self._init_train()
  File "/home/xuexidi/xue/embclip-habitat/habitat_baselines/rl/ppo/ppo_trainer.py", line 248, in _init_train
    self._init_envs()
  File "/home/xuexidi/xue/embclip-habitat/habitat_baselines/rl/ppo/ppo_trainer.py", line 201, in _init_envs
    workers_ignore_signals=is_slurm_batch_job(),
  File "/home/xuexidi/xue/embclip-habitat/habitat_baselines/utils/env_utils.py", line 107, in construct_envs
    workers_ignore_signals=workers_ignore_signals,
  File "/home/xuexidi/xue/embclip-habitat/habitat/core/vector_env.py", line 195, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/home/xuexidi/xue/embclip-habitat/habitat/core/vector_env.py", line 195, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/home/xuexidi/xue/embclip-habitat/habitat/core/vector_env.py", line 97, in __call__
    res = self.read_fn()
  File "/home/xuexidi/xue/embclip-habitat/habitat/utils/pickle5_multiprocessing.py", line 68, in recv
    buf = self.recv_bytes()
  File "/home/xuexidi/anaconda3/envs/embclip-habitat/lib/python3.6/multiprocessing/connection.py", line 216, in recv_bytes
    buf = self._recv_bytes(maxlength)
  File "/home/xuexidi/anaconda3/envs/embclip-habitat/lib/python3.6/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/home/xuexidi/anaconda3/envs/embclip-habitat/lib/python3.6/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer`

CwwwwS, 2022-11-07:
I have the same problem, do you know how to solve it?

xuexidi, 2022-11-07:
> I have the same problem, do you know how to solve it?

please see this:
https://github.com/allenai/embodied-clip/issues/6

CwwwwS, 2022-11-08:
Thank you for your reply, I checked this link above, but still did not find a suitable solution. The problem I encountered is very inexplicable, the previous night can still run, but the next day will not work, is it because I reinstalled ROS or opencv, but I think it does not affect my conda environment. I am using a single GPU

CwwwwS, 2022-11-08:
I've solved it, it's a GPU driver problem.  Thank you very much.

closed 2023-01-26 by rpartsey

taylorlulu, 2023-09-19:
> I've solved it, it's a GPU driver problem. Thank you very much.

@CwwwwS I faced the same problem, may I ask how you solved this problem?