facebookresearch/habitat-lab #917: About distribution of losses

RichterLee, 2022-07-29:
## Habitat-Lab
Habitat-Lab: master, latest version
## ❓ Questions
[image]

Hi, I have a question about the ddppo algorithm in /habitat_baselines/rl/ppo/ppo_trainer.py. I find that the losses are first sorted, then distrib.all_reduce is called, and then losses divided by world_size. Is there a reason why losses are first sorted here? If I want to add additional losses,  can I just write it as following?

stats = torch.tensor(
                [losses[k] for k in loss_name_ordering] + [additional loss] + [count_steps_delta],
                device="cpu",
                dtype=torch.float32,
            )

Thank you so much!

dhruvbatra, 2022-07-30:
@erikwijmans

erikwijmans, 2022-08-01:
The sort is so that all ranks have the same order of keys (prior to python 3.7, dictionaries weren't ordered by default so `.keys()` would return the keys in an arbitrary order). If it wasn't the same, we may end up reducing the value loss with the action loss (or something similarly incorrect). 

When reducing additional losses, you need to guarantee that all ranks will append their additional losses in the same order, with 1 loss that is guaranteed.

RichterLee, 2022-08-01:
> @erikwijmans  Thank you so much! If I'm understanding it correctly, it's the problem with the dictionaries. So if I don't use dictionary to store these loss, and explicitly list all the losses, there won't be any problem with it, right? 

For example,

def _coalesce_post_step(self, loss_1, loss_2, loss_3):
    stats = torch.tensor([loss_1, loss_2, loss_3], device="cpu")
    stats = self._all_reduce(stats)

erikwijmans, 2022-08-01:
Yep.