facebookresearch/habitat-lab #918: About .glb files

EchoTHChen, 2022-08-01:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.2 

Habitat-Sim: v0.6.2

## ❓ Questions and Help
When I download the matterport3d dataset, I did not find out '.glb' files. I use the following script to download:
`python download_mp.py --task habitat -o data/scene_datasets/mp3d/`. How could I get the glb files? (Besides, from which link I could get the .glb only? )

closed 2022-08-01 by EchoTHChen

ylqi, 2022-11-23:
Hi, I have the same question. Those downloaded *.zip files only include files such as *.png, *.sens, *.ply, *.json or *.cam. Did you solve this question?

EchoTHChen, 2022-11-23:
@ylqi 
 I solved it by adding an option:
`python download_mp.py --task habitat -o path/to/download/`

EchoTHChen, 2022-11-23:
Note that this download script requires python 2.7 to run.

EchoTHChen, 2022-11-23:
@ylqi . Maybe you should remove all the downloaded matterport3d data and re-download.

EchoTHChen, 2022-11-23:
The download path you set should be empty

ylqi, 2022-11-23:
Thank you! 
I find the `*.glb` in `tasks/mp3d_habitat.zip`.
(I spent a lot of time downloading the `scans/` folder, so I thought it would be under the `scan/` folder.)