facebookresearch/habitat-lab #931: FileNotFoundError during SAVI training

sivannavis, 2022-08-23:
This is still the case when I evaluate the pretrained model using `py ss_baselines/savi/run.py --run-type eval --exp-config ss_baselines/savi/config/semantic_audionav/savi.yaml EVAL_CKPT_PATH_DIR data/pretrained_weights/semantic_audionav/savi/best_val.pth EVAL.SPLIT test USE_SYNC_VECENV True RL.DDPPO.pretrained False`

After some debugging I noticed `config.TASK_CONFIG` is good before the `make_env_fn` function. And it seems it's this part in initializing the `Env` Class that changes the `SCENE_DATASET` to "default", which I'm a bit confused about: https://github.com/facebookresearch/habitat-lab/blob/380ac0a7d8c4ead1532f109b15d329473212eae9/habitat/core/env.py#L95

> 
> self._config.SIMULATOR.SCENE_DATASET = (
>                 self.current_episode.scene_dataset_config
>             )

https://github.com/facebookresearch/habitat-lab/blob/380ac0a7d8c4ead1532f109b15d329473212eae9/habitat/core/dataset.py#L60

> scene_dataset_config: str = attr.ib(
>         default="default", validator=not_none_validator
>     )
> 

Could you help me take a look and point me on the right way?

_Originally posted by @sivannavis in https://github.com/facebookresearch/sound-spaces/issues/89#issuecomment-1223389426_

sreeharshaparuchur1, 2022-09-07:
Hey @sivannavis !

I'm facing a similar error while trying to run the av_wan baseline from the soundspaces repo. I too have traced this absurd behaviour of the path changing to 'default' instead of 'mp3d' (replica, in my case) down to this part in the code. May I know if you have figuired a work around for the same that does not have an effect on the way habitat-sim goes about its task.

I am using habitat2.2, are you using the same version?

sivannavis, 2022-09-07:
@sreeharshaparuchur1 Hi! Yes, I'm using the same version 2.2, and my workaround is just simply commenting this part that changes the scene_dataset string, in this thread https://github.com/facebookresearch/sound-spaces/issues/89#issuecomment-1233326255 .

sreeharshaparuchur1, 2022-09-09:
@sivannavis Ah, yeah I did that too. It seems to work alright, even with other baselines. It does beg the question as to why it was added in the first place. Also, are you running code with soundspaces 2.0 or soundspaces 1.0?

sivannavis, 2022-09-09:
I think I'm running soundspaces 1.0. I do have the same question though...

sivannavis, 2022-09-13:
After some checking, it turns out to be a version mismatch between Habitat-lab and SAVI code. I rolled back to version 0.2.1 and problem solved.

closed 2022-09-13 by sivannavis