facebookresearch/habitat-lab #947: ERROR in running examples/benchmark.py

keeper1314, 2022-09-11:
The error traceback is :
Traceback (most recent call last):
  File "examples/benchmark.py", line 10, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/data1/fuyu/habitat-lab/habitat/sims/habitat_simulator/actions.py", line 12, in <module>
    import habitat_sim
  File "/data1/fuyu/.conda/envs/Object_Goal_Navigation/lib/python3.8/site-packages/habitat_sim-0.1.5-py3.8-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/data1/fuyu/.conda/envs/Object_Goal_Navigation/lib/python3.8/site-packages/habitat_sim-0.1.5-py3.8-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: /data1/fuyu/.conda/envs/Object_Goal_Navigation/lib/python3.8/site-packages/habitat_sim-0.1.5-py3.8-linux-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.cpython-38-x86_64-linux-gnu.so: undefined symbol: __cpu_model

I don't know how to fix it and can't find any solution in other issue.
I'm using this repository: https://github.com/devendrachaplot/Object-Goal-Navigation
with habitat_sim-0.1.5, pytorch==1.6.0, torchvision==0.7.0, cudatoolkit==10.2.

aclegg3, 2022-09-12:
@keeper1314 this looks like a compatibility issue with gcc compiler. Did you install habitat-sim-0.1.5 from conda or build from source? What gcc version is installed on your system?

keeper1314, 2022-09-14:
I build habitat-sim-0.1.5 from source. gcc version is 9.1.0.