facebookresearch/habitat-lab #960: Incorrect semantic mask accquired from habitat-api

WangJuan6, 2022-09-28:
Hi:

I used the habitat-api to get the semantic mask in the ARNzJeq3xxb scene of the mp3d dataset:

In the reset() function:
```
scene = self.habitat_env.sim.semantic_annotations()
instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
self.mapping = np.array([ instance_id_to_label_id[i] for i in range(len(instance_id_to_label_id)) ])
semantic_mask_gt = np.take(self.mapping, obs['semantic'])
```

In the step() function:
```
semantic_mask_gt = np.take(self.mapping, obs['semantic'])
```

I save the  mask of the plant using the code below:
```
semantic_mask_plant = np.ones(semantic_mask_gt.shape) * 255
semantic_mask_plant[semantic_mask_gt==14] = 0 # plant
cv2.imwrite("./mask_plant_" + str(self.timestep).zfill(6) + ".png",semantic_mask_plant)
```
However I noticed the semantic mask of the plant has the wrong region, 

The RGB image and semantic mask of the plant at timestep 19 are shown below:

[image]

[image]

The RGB image and the semantic mask of the plant at timestep 196 are shown below:

[image]

[image]

The semantic mask at timestep 19 is totally wrong, and the semantic mask at timestep 196 has both the incorrect and the correct region.

I want to know how to get the totally correct semantic mask. 
Can you help me to solve this problem?

Thanks

dhruvbatra, 2022-09-29:
Not commenting on this specific case (because I can't visually understand what's going on), but in general, MP3D annotations are known to be noisy (and the simulator cannot change that; noise in, noise out). Try HM3D-Semantics (https://aihabitat.org/datasets/hm3d-semantics/); still not noise-free, but better. If you still see such gross errors in HM3D-Semantics, then it's worth looking at your code.

WangJuan6, 2022-09-29:
Thanks for your reply, I'll try the HM3D-Semantics dataset.

closed 2022-09-29 by WangJuan6