facebookresearch/habitat-lab #985: I can't run examples

MichaelGaliciaZ, 2022-10-25:
I installed habitat-sim correctly but when I install habitat-lab and try an example with:

(habitat_NB_with_habitat_lab) miguel@miguel:~/habitat-lab$ python examples/example_pointnav.py 
Traceback (most recent call last):
  File "examples/example_pointnav.py", line 7, in <module>
    import habitat
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/core/benchmark.py", line 18, in <module>
    from habitat.core.env import Env
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/core/env.py", line 22, in <module>
    from habitat.tasks import make_task
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/__init__.py", line 7, in <module>
    from habitat.tasks.registration import make_task  # noqa: F401
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/registration.py", line 28, in <module>
    _try_register_rearrange_task()
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/rearrange/__init__.py", line 13, in _try_register_rearrange_task
    import habitat.tasks.rearrange.actions
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/rearrange/actions.py", line 21, in <module>
    from habitat.tasks.rearrange.grip_actions import (
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/rearrange/grip_actions.py", line 14, in <module>
    from habitat.tasks.rearrange.rearrange_sim import RearrangeSim
  File "/home/miguel/anaconda3/envs/habitat_NB_with_habitat_lab/lib/python3.7/site-packages/habitat/tasks/rearrange/rearrange_sim.py", line 36, in <module>
    from habitat_sim.robots import FetchRobot, FetchRobotNoWheels
ModuleNotFoundError: No module named 'habitat_sim.robots'

Can you help me with that please?

aclegg3, 2022-10-26:
Hey @MichaelGaliciaZ, 

The `robots` module was moved from habitat-sim to habitat-lab, so my guess would be you've got some outdated or mis-matched versions here. 

habitat-sim `main` or nightly conda corresponds to habitat-lab `main`. Otherwise you should be using equivalently tagged versions (e.g. both `stable` or both `vx.x`).

MichaelGaliciaZ, 2022-10-27:
I also thought that habitat-sim **nightly** corresponded to habitat-lab --branch **main**, but when I installed it, it did not work. I partially solved this problem by installing habitat-lab --branch **header_changes** 

**After testing the examples, run:**

 (habitat_NB_Header_Changes) miguel@miguel:~/habitat-lab$ python examples/interactive_play.py 

Everything seemed to be going well, until finally the following error was shown:

[19:40:07:381725]:[Physics] PhysicsManager.cpp(287)::addObject : Simplified template handle : 009_gelatin_box  | newObjectHandle : 009_gelatin_box_:0000
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  97
  Current serial number in output stream:  97

**The other examples work properly.**

I appreciate your answers, I will try to reinstall habitat-sim and habitat-lab from main(stable or nightly), but if there is a way to solve the previous error it will be useful.

aclegg3, 2022-10-28:
@MichaelGaliciaZ, glad you got the examples worked out for the most part.

`interactive_play.py` is unique here in that it uses pygame to run an interactive application window using your display system. In this case, my best guess is that the issue is coming from your system's GLX configuration. Maybe a driver issue?

Tagging @mosra for graphics driver insight and @ASzot for interactive_play.py

alper111, 2022-11-30:
I have `habitat=0.2.2`and `habitat-sim=0.2.2` (installed from conda, not nightly), and I have this error, too.

JalongMa, 2022-12-01:
@alper111 
I encountered the same problem as you. So I recompiled the source of habitat-sim with **tag 0.2.2**, then habitat-sim=0.2.2 works well. 
There are something wrong with conda package of `habitat-sim=0.2.2`. It's inconsistent with corresponding source code.

aclegg3, 2022-12-01:
@alper111 @JalongMa sorry for this. I think it was an issue with our version release workflow which created new conda packages for stable 0.2.2 from current `main` branch. I've removed these, so the issue should be resolved.

aclegg3, 2022-12-01:
Given that the original issue is stale, I'm closing this thread.

closed 2022-12-01 by aclegg3

yihe-61wu, 2023-01-09:
> @alper111 @JalongMa sorry for this. I think it was an issue with our version release workflow which created new conda packages for stable 0.2.2 from current `main` branch. I've removed these, so the issue should be resolved.

I was not having the same problem as the original post, although I couldn't run the examples either. I fixed the problem by conda install habitat-sim=0.2.3 -- Despite the most updated release of 0.2.3, when conda install thing following the most recommended lines in readme, I installed 0.2.2.