facebookresearch/habitat-lab #1015: How to generate position images when testing at habitat-baseline

yingchengyang, 2022-11-23:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim: master

## ❓ Questions and Help

Thanks a lot! I have found it in the tutorial on Hab-Sim basics and the file habitat-lab/habitat/utils/visualizations/maps.py for using it in Habitat-lab. But I unfortunately do not find how to draw this map when evaluating trained models by using the code in habitat-baselines. As for habitat-baselines, are there any parameters in the config to draw this map to the video?

Thank you again for your patient reply!

dhruvbatra, 2022-11-23:
Yeah, looks like we are lacking a clear tutorial on how to make such a video. 

Check out this visualization code from @Ram81: 
https://github.com/Ram81/habitat-imitation-baselines/blob/master/examples/objectnav_replay.py

And this test that will tell you how to combine everything: 
https://github.com/Ram81/habitat-imitation-baselines/blob/master/test/test_visual_utils.py

yingchengyang, 2022-11-24:
Thanks a lot for your valuable and patient comment. Based on this test [https://github.com/Ram81/habitat-imitation-baselines/blob/master/test/test_visual_utils.py](https://github.com/Ram81/habitat-imitation-baselines/blob/master/test/test_visual_utils.py), I successfully add the map of the agent's route in the video! However, I still can not draw the target point and the shortest path (target solution) to the video. As I find that there are parameters _C.habitat.task.top_down_map.draw_shortest_path and _C.habitat.task.top_down_map.draw_goal_positions, the info I get by env.step() does not seem to contain target information. So how can I draw the target point and the shortest path (target solution) to the video?

Thank you so much for patient reply!

Ram81, 2022-12-22:
Hi @yingchengyang,

You can draw the target object point by setting [`draw_goal_positions=True`](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/config/default_structured_configs.py#L406) and draw the shortest path by setting [`draw_shortest_path=True`](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/config/default_structured_configs.py#L404).

To fetch the top down map at each step you can access the `infos["top_down_map"]` key. To enable top down map just pass `top_down_map` measure in the measurements [list here](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/config/habitat/task/objectnav.yaml#L11). 

You can also refer to[ this example script](https://github.com/Ram81/habitat-imitation-baselines/blob/master/examples/objectnav_replay.py) I have in my project which generates a video with RGB + top down maps observations.

rpartsey, 2023-01-09:
@yingchengyang The tutorial on TopDownMap visualization has just been merged to main.
Please, use this [notebook](https://github.com/facebookresearch/habitat-lab/blob/main/examples/tutorials/colabs/Habitat_Lab_TopdownMap_Visualization.ipynb) or corresponding [python file](https://github.com/facebookresearch/habitat-lab/blob/main/examples/tutorials/nb_python/Habitat_Lab_TopdownMap_Visualization.py) as a reference.

closed 2023-01-09 by rpartsey