facebookresearch/habitat-lab #1043: ImportError: cannot import name 'orthonormalize_rotation_shear' from 'habitat_sim.utils.common'

maytusp, 2022-12-17:
Hi, I installed habitat following https://github.com/facebookresearch/habitat-lab/.
But it occurred this error.

## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.2.3
Habitat-Sim: 0.2.1

How could I fix the issue?

vincentpierre, 2023-01-04:
You do need to have the same versions for habitat-lab and for habitat-sim. Some methods are moved or renamed which can causes issues like the one you are facing. If you find such an error while using the same version, please let us know.
If you want to use the latest development versions, you need to be on main for habitat-lab and use `-c aihabitat-nightly` when installing habitat-sim.

To install the latest version (0.2.3) of habitat sim with the main branch of habitat-lab, please run :
```
conda create -n habitat_023 python=3.7 cmake=3.14.0 -y
conda activate habitat_023
conda install habitat-sim withbullet -c conda-forge -c aihabitat -y
pip install -e habitat-lab 
python examples/example.py
```
Let us know if this works.

In the future, please report all installation commands you ran and the whole stack trace of the error you encountered. This will help us reproduce your issues!

maytusp, 2023-01-07:
It works. Thank you.

closed 2023-01-07 by maytusp