facebookresearch/habitat-lab #1051: How can I define my own action?

GuoPingPan, 2022-12-25:
# 😢 I want to define my own environment and actions.

## First: I create my own action just like below
```
@attr.s(auto_attribs=True, slots=True)
class CustomActuationSpec:
    action: int

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyForward(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyRight(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyLeft(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat.registry.register_action_space_configuration(name='Custom')
class CustomActionSpaceConfiguration(HabitatSimV0ActionSpaceConfiguration):
    def get(self):
        config = super().get()

        print('sign up action: \n')
        print(HabitatSimActions.NOISY_FORWARD)
        print(HabitatSimActions.NOISY_RIGHT)
        print(HabitatSimActions.NOISY_LEFT)

        config[HabitatSimActions.NOISY_FORWARD] = habitat_sim.ActionSpec(
            "noisy_forward",
            CustomActuationSpec(0),
        )
        config[HabitatSimActions.NOISY_RIGHT] = habitat_sim.ActionSpec(
            "noisy_right",
            CustomActuationSpec(1),
        )
        config[HabitatSimActions.NOISY_LEFT] = habitat_sim.ActionSpec(
            "noisy_left",
            CustomActuationSpec(2),
        )

        return config

```

## Second: In order to use it, I extend the action space
```
 HabitatSimActions.extend_action_space("NOISY_FORWARD")  # 16
        HabitatSimActions.extend_action_space("NOISY_RIGHT")    # 17
        HabitatSimActions.extend_action_space("NOISY_LEFT")     # 18

        with habitat.config.read_write(config_env):
            config_env.habitat.simulator.action_space_config = 'Custom'
```
## Third: But I find it don't work, because the root of Env.step is task.step. 
```
    def step(
        self, action: Union[int, str, Dict[str, Any]], **kwargs
    ) -> Observations:
        assert (
            self._episode_start_time is not None
        ), "Cannot call step before calling reset"
        assert (
            self._episode_over is False
        ), "Episode over, call reset before calling step"
        assert (
            not self._episode_force_changed
        ), "Episode was changed either by setting current_episode or changing the episodes list. Call reset before stepping the environment again."

        # Support simpler interface as well
        if isinstance(action, (str, int, np.integer)):
            action = {"action": action}

        observations = self.task.step(
            action=action, episode=self.current_episode
        )

        self._task.measurements.update_measures(
            episode=self.current_episode,
            action=action,
            task=self.task,
            observations=observations,
        )

        self._update_step_stats()

        return observations
```
### So I modify the config as below
```
        with habitat.config.read_write(config_env):
            config_env.habitat.simulator.action_space_config = 'Custom'
            from omegaconf.dictconfig import DictConfig
            print(config_env.habitat.task.actions)
            config_env.habitat.task.actions.setdefault('noisy_forward', DictConfig({'type': 'NoisyForward'}))
            config_env.habitat.task.actions.setdefault('noisy_right',  DictConfig({'type': 'NoisyRight'}))
            config_env.habitat.task.actions.setdefault('noisy_left', DictConfig({'type': 'NoisyLeft'}))

```

## 🐛  Fourth: Finally, here comes an error: `omegaconf.errors.ValidationError: Invalid type assigned: dict is not a subclass of ActionConfig. value: {'type': 'NoisyForward'}` 

I debug the `EmbodiedTask._init_entities` and find that the `entity_cfg` has a value `__metadata` as below

[image]

And the diffenerce between my own defined action `NoisyForward` and  the default  action `turn_left` is that my action's `__metadata.object_type = dict`, and come the error `omegaconf.errors.ValidationError: Invalid type assigned: dict is not a subclass of ActionConfig. value: {'type': 'NoisyForward'}` .

## Last: I search the `default_structured_configs.py` and try to add the below code to deal with it, but doesn't work
```
from hydra.core.config_store import ConfigStore
from habitat.config.default_structured_configs import ActionConfig
from dataclasses import dataclass

cs = ConfigStore.instance()

@dataclass
class NoisyForwardActionConfig(ActionConfig):
    type: str = 'NoisyForwardAction'

cs.store(
    package="habitat.task.actions.noisy_forward",
    group="habitat/task/actions",
    name="noisy_forward",
    node=NoisyForwardActionConfig,
)
```

## 🆘   Now I am exhausted，why it is so hard to define my own action. If you can offer me some help, I will be very thankful!!! 🌹

GuoPingPan, 2022-12-27:
@vincentpierre Can you help me to check about it?

vincentpierre, 2023-01-04:
Hi, have you had a look at this page ? https://github.com/facebookresearch/habitat-lab/blob/main/examples/new_actions.py It gives an example on how to create your own action space. It is fairly hard to find I must admit.
Could you provide some usable code samples with your issues, I had a hard time piecing all your code snippets into code I could run.

I tried running this code on `main` and it seems to work fine for me : 

```python
import attr
import numpy as np

import habitat
import habitat_sim
from habitat.config.default_structured_configs import ActionConfig
from habitat.sims.habitat_simulator.actions import (
    HabitatSimActions,
    HabitatSimV1ActionSpaceConfiguration,
    HabitatSimV0ActionSpaceConfiguration,
)
from habitat.tasks.nav.nav import SimulatorTaskAction

def _noisy_action_impl(scene_node, actuation_spec):
    print(f"_noisy_action_impl called with {scene_node}, {actuation_spec}")

@attr.s(auto_attribs=True, slots=True)
class CustomActuationSpec:
    action: int

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyForward(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyRight(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat_sim.registry.register_move_fn(body_action=True)
class NoisyLeft(habitat_sim.SceneNodeControl):
    def __call__(
            self,
            scene_node: habitat_sim.SceneNode,
            actuation_spec: CustomActuationSpec,
    ):
        _noisy_action_impl(
            scene_node,
            actuation_spec.action,
        )

@habitat.registry.register_task_action
class NoisyForwardAction(SimulatorTaskAction):
    def _get_uuid(self, *args, **kwargs) -> str:
        return "noisy_forward"

    def step(self, *args, **kwargs):
        return self._sim.step(HabitatSimActions.NOISY_FORWARD)

@habitat.registry.register_task_action
class NoisyLeftAction(SimulatorTaskAction):
    def _get_uuid(self, *args, **kwargs) -> str:
        return "noisy_left"

    def step(self, *args, **kwargs):
        return self._sim.step(HabitatSimActions.NOISY_LEFT)

@habitat.registry.register_task_action
class NoisyRightAction(SimulatorTaskAction):
    def _get_uuid(self, *args, **kwargs) -> str:
        return "noisy_right"

    def step(self, *args, **kwargs):
        return self._sim.step(HabitatSimActions.NOISY_RIGHT)

@habitat.registry.register_action_space_configuration(name='Custom')
class CustomActionSpaceConfiguration(HabitatSimV0ActionSpaceConfiguration):
    def get(self):
        config = super().get()

        print('sign up action: \n')
        print(HabitatSimActions.NOISY_FORWARD)
        print(HabitatSimActions.NOISY_RIGHT)
        print(HabitatSimActions.NOISY_LEFT)

        config[HabitatSimActions.NOISY_FORWARD] = habitat_sim.ActionSpec(
            "noisy_forward",
            CustomActuationSpec(0),
        )
        config[HabitatSimActions.NOISY_RIGHT] = habitat_sim.ActionSpec(
            "noisy_right",
            CustomActuationSpec(1),
        )
        config[HabitatSimActions.NOISY_LEFT] = habitat_sim.ActionSpec(
            "noisy_left",
            CustomActuationSpec(2),
        )

        return config

def main():
    HabitatSimActions.extend_action_space("NOISY_FORWARD")  # 16
    HabitatSimActions.extend_action_space("NOISY_RIGHT")    # 17
    HabitatSimActions.extend_action_space("NOISY_LEFT")     # 18

    config_env = habitat.get_config(
        config_path="benchmark/nav/pointnav/pointnav_habitat_test.yaml"
    )

    with habitat.config.read_write(config_env):
        # Add a simple action config to the config.habitat.task.actions dictionary
        config_env.habitat.task.actions["NOISY_FORWARD"] = ActionConfig(
            type="NoisyForwardAction"
        )
        config_env.habitat.task.actions["NOISY_RIGHT"] = ActionConfig(
            type="NoisyRightAction"
        )
        config_env.habitat.task.actions["NOISY_LEFT"] = ActionConfig(
            type="NoisyLeftAction"
        )

        config_env.habitat.simulator.action_space_config = 'Custom'

    with habitat.Env(config=config_env) as env:
        env.reset()
        env.step("NOISY_FORWARD")
        env.step("NOISY_RIGHT")

if __name__ == "__main__":
    main()

```

But I am not sure it is what you were trying to do. One subtlety is that I have to register a move function AND a task action for it to work. Also, if there is a typo in the name of the action, the code will return an error that is not intuitive at all. I think this might be part of the issue.
Also note that 

```python
config_env.habitat.task.actions["NOISY_FORWARD"] = ActionConfig(
            type="NoisyForwardAction"
        )
```
needs to be an ActionConfig and cannot be a regular dictionary since we moved to Hydra.
I hope this helps and sorry for the long silence over the holidays.

closed 2023-01-05 by GuoPingPan