facebookresearch/habitat-lab #1073: Cannot test PointNav example, with the error 'omegaconf.errors.ConfigAttributeError: ListConfig does not support attribute access'

Yuchen-Song, 2023-01-10:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.3 (stable)

Habitat-Sim: v0.2.2 (stable)

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. Install habitat-sim (headless & bullet) through anaconda, as stated in guidelines `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`
2. pip install habitat-lab stable version `pip install -e habitat-lab`
3. Test the pointNav task `python examples/example_pointnav.py`

## Expected behavior

Error messages are: 

> Traceback (most recent call last):
>   File "/home/[MyUserName]/habitat-lab/examples/example_pointnav.py", line 30, in <module>
>     example()
>   File "/home/[MyUserName]/habitat-lab/examples/example_pointnav.py", line 19, in example
>     observations = env.reset()  # noqa: F841
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/core/env.py", line 260, in reset
>     self.reconfigure(self._config)
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/core/env.py", line 345, in reconfigure
>     self._sim.reconfigure(self._config.simulator)
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 451, in reconfigure
>     self._update_agents_state()
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 395, in _update_agents_state
>     agent_id,
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 593, in set_agent_state
>     agent.set_state(new_state, reset_sensors)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 255, in set_state
>     self.body.object.rotation = quat_to_magnum(state.rotation)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/utils/common.py", line 49, in quat_to_magnum
>     return mn.Quaternion(quat.imag, quat.real)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/listconfig.py", line 185, in __getattr__
>     "ListConfig does not support attribute access"
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/base.py", line 237, in _format_and_raise
>     type_override=type_override,
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/_utils.py", line 899, in format_and_raise
>     _raise(ex, cause)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/_utils.py", line 797, in _raise
>     raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
> omegaconf.errors.ConfigAttributeError: ListConfig does not support attribute access
>     full_key: habitat.simulator.agents.main_agent.start_rotation[imag]
>     reference_type=List[float]
>     object_type=list

## Additional context

The whole output in the console: 

> /home/[MyUserName]/anaconda3/envs/habitat/bin/python /home/[MyUserName]/habitat-lab/examples/example_pointnav.py 
> pybullet build time: Sep 22 2020 00:55:20
> 2023-01-10 16:04:12,463 Initializing dataset PointNav-v1
> 2023-01-10 16:04:12,706 initializing sim Sim-v0
> [16:04:12:711116]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
> [16:04:12:711208]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DSolid:capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
> [16:04:12:711233]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DWireframe:capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
> [16:04:12:711251]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneSolid:coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
> [16:04:12:711265]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneWireframe:coneWireframe_segments_32_halfLen_1.25) created and registered.
> [16:04:12:711278]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeSolid:cubeSolid) created and registered.
> [16:04:12:711295]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeWireframe:cubeWireframe) created and registered.
> [16:04:12:711311]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderSolid:cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
> [16:04:12:711326]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderWireframe:cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
> [16:04:12:711341]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereSolid:icosphereSolid_subdivs_1) created and registered.
> [16:04:12:711376]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereWireframe:icosphereWireframe_subdivs_1) created and registered.
[... 85 lines cut ...]
>     agent_id,
>   File "/home/[MyUserName]/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 593, in set_agent_state
>     agent.set_state(new_state, reset_sensors)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 255, in set_state
>     self.body.object.rotation = quat_to_magnum(state.rotation)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/habitat_sim/utils/common.py", line 49, in quat_to_magnum
>     return mn.Quaternion(quat.imag, quat.real)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/listconfig.py", line 185, in __getattr__
>     "ListConfig does not support attribute access"
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/base.py", line 237, in _format_and_raise
>     type_override=type_override,
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/_utils.py", line 899, in format_and_raise
>     _raise(ex, cause)
>   File "/home/[MyUserName]/anaconda3/envs/habitat/lib/python3.7/site-packages/omegaconf/_utils.py", line 797, in _raise
>     raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
> omegaconf.errors.ConfigAttributeError: ListConfig does not support attribute access
>     full_key: habitat.simulator.agents.main_agent.start_rotation[imag]
>     reference_type=List[float]
>     object_type=list
> [16:04:15:674960]:[Scene] SemanticScene.h(47)::~SemanticScene : Deconstructing SemanticScene
> [16:04:15:691945]:[Gfx] Renderer.cpp(72)::~Impl : Deconstructing Renderer
> [16:04:15:692005]:[Gfx] WindowlessContext.h(17)::~WindowlessContext : Deconstructing WindowlessContext
> [16:04:15:941496]:[Sim] Simulator.cpp(66)::~Simulator : Deconstructing Simulator
> 
> Process finished with exit code 1

Yuchen-Song, 2023-01-10:
My `omegaconf` version is 2.3.0

vincentpierre, 2023-01-11:
Hi @Yuchen-Song , 
I was unable to reproduce this on my machine on the latest v0.2.3 version.
Did you recently install habitat-sim? It looks like your habitat-sim and habitat-lab versions are not the same. Could you try reinstalling habitat-sim to the latest v0.2.3?
Thank you.

Yuchen-Song, 2023-01-12:
Thank you, @vincentpierre 
When I reinstalled the 0.2.3 version of habitat-sim, the probelm no longer occurs.
BTW, the default install command for habitat-sim gets 0.2.2, for anyone who may face the same problem, should specify the version habitat-sim==0.2.3 (or the same version as habitat-lab) to avoid this.

closed 2023-01-12 by Yuchen-Song