facebookresearch/habitat-lab #1074: Habitat.config error in Google Colab

JaechangKo, 2023-01-10:
For this tutorial session: https://www.youtube.com/watch?v=KbbfD7RlJLM&list=PLGywud_-HlCORC0c4uj97oppQrGiB6JNy
When I run the code 'Path Setup and Imports', habitat.config error doesn't allow me to move forward. Is this happened because of the incompatibility of TensorFlow version?

ImportError: cannot import name 'Config' from 'habitat.config' (/content/habitat-lab/habitat-lab/habitat/config/init.py)

Can some one let me know how to deal with this??
[image]

rpartsey, 2023-01-26:
Hi @JaechangKo check out [this issue](https://github.com/facebookresearch/habitat-sim/issues/1818). There is a known problem with Colab. Try the solution proposed there and let me know if there are more issues.

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.