facebookresearch/habitat-lab #1087: EpisodicCompassSensor gives incorrect observation

jiaming-robot-learning, 2023-01-20:
## Habitat-Lab versions
Habitat-Lab: main branch

## 🐛 Bug
The EpisodicCompassSensor defined in [habitat-lab/habitat/tasks/nav/py](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/tasks/nav/nav.py#L394) is not correct.
```python
def get_observation(
        self, observations, episode, *args: Any, **kwargs: Any
    ):
        agent_state = self._sim.get_agent_state()
        rotation_world_agent = agent_state.rotation
        rotation_world_start = quaternion_from_coeff(episode.start_rotation)

        if isinstance(rotation_world_agent, quaternion.quaternion):
            return self._quat_to_xy_heading(
                rotation_world_agent.inverse() * rotation_world_start # should be rotation_world_start.inverse() * rotation_world_agent
            )
        else:
            raise ValueError("Agent's rotation was not a quaternion")
```
the line 
> __rotation_world_agent.inverse() * rotation_world_start__ 

should actually be 

>__rotation_world_start.inverse() * rotation_world_agent__

because we want to convert the current agent's states in the world coordinate to the reference frame defined by the start state of the current episode.

jiaming-robot-learning, 2023-01-20:
Sorry it's not a bug. It's the same reason as I commented in #1086

closed 2023-01-20 by jiaming-robot-learning

dhruvbatra, 2023-01-20:
@jiaming-robot-learning -- would you consider sending a PR w/ a docstring/explanation at the right place in the code so others don't make the same mistake?

jiaming-robot-learning, 2023-01-21:
I'd love to! But I'm still not sure if this is correct.

From my understanding, this line
```
rotation_world_agent.inverse() * rotation_world_start
```
transforms the world (0,0,-1) to the robot's local frame, then it calls the following function:
```python
def _quat_to_xy_heading(self, quat):
        direction_vector = np.array([0, 0, -1])

        heading_vector = quaternion_rotate_vector(quat, direction_vector)

        phi = cartesian_to_polar(-heading_vector[2], heading_vector[0])[1]
        return np.array([phi], dtype=np.float32)
```
Notice
> phi = cartesian_to_polar(-heading_vector[2], heading_vector[0])[1]

first rotate the vector - pi/2, then calculate the angle between the rotated vector and x-axis. By doing this, it actually returns __$\pi - \theta$__, but not theta, which is the actual angle rotated by the robot.

Can anyone familiar with this explain it?