facebookresearch/habitat-lab #1089: Failed to import habitat_sim following tutorial ECCV 2020: Navigation

Yuhanelle, 2023-01-20:
Hi, I followed the turotial ECCV 2020: Navigation in colab and I always receive this error when I run the code. I tried to change the path, but it didn't work. 

[image]

Can anyone help me with this? Thank you in advance.

rpartsey, 2023-01-26:
Hi @Yuhanelle check out [this issue](https://github.com/facebookresearch/habitat-sim/issues/1818). There is a known problem with Colab. Try the solution proposed there and let me know if there are more issues.

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.

closed 2024-01-16 by aclegg3