facebookresearch/habitat-lab #1115: Should `VelocityControlActionConfig` be added to the ConfigStore

luisenp, 2023-02-02:
[This](https://github.com/facebookresearch/habitat-lab/blob/f64ca16901947947cb5a74ede34e5325f2b9f4fc/habitat-lab/habitat/config/default_structured_configs.py#L156) config option doesn't seem to be added to the config store, is this deliberate? I was trying to do something like the following:
```yaml
@package _global_
defaults:
  - /habitat/task/actions:
    - velocity_control
```
However, I had to add the following to the default configs to make this work (I didn't see any existing entries for this config option)
```python
cs.store(
    package="habitat.task.actions.velocity_control",
    group="habitat/task/actions",
    name="velocity_control",
    node=VelocityControlActionConfig,
)
```
Do let me know if this is not the intended way for this to be used. Thanks.

cc @jimmytyyang @KRobG

jimmytyyang, 2023-02-02:
Thanks for reporting this. I think this is not intended

luisenp, 2023-02-03:
In case it's helpful, I made a PR with the fix I proposed above.

PR #1119 (2023-02-16): Added velocity_control action to ConfigStore.
Closes #1115
closed 2023-02-16 by jimmytyyang