facebookresearch/habitat-lab #1124: queue.Full error when training objectnav with VER

sanjayharesh, 2023-02-08:
Hi, 

I am trying to train an ObjectNav agent on floorplanner data with VER with a clip backbone but keep getting the error attached below. I am using 4 A100 GPUs with 32 environments. [Here's](https://textdoc.co/bGu4laATK1I3WdjE) the config file for additional context.

habitat-lab: v0.2.3 ([specific fork](https://github.com/3dlg-hcvc/habitat-lab/tree/fp-clean))
habitat-sim: v0.2.3 ([specific fork](https://github.com/3dlg-hcvc/habitat-sim/tree/e3842efc478b50aa1cdd4623b2888d5232edb9a4))

### Error Message:
The training fails after some time with the following error. Note that it happens at random/unpredictable times.
Also since the error message is too cryptic so not sure what other information is relevant. Let me know if I can add more details here?

> 2023-02-08 01:31:59,854 update: 2030	fps: 1392.4	window fps: 1447.1	frames: 12356606
> 2023-02-08 01:31:59,854 Average window size: 6400  reward: 0.733  distance_to_goal: 5.200  success: 0.048  spl: 0.026  soft_spl: 0.193  distance_to_goal_reward: -0.000
> 2023-02-08 01:31:59,871 env: step env: 16.4ms  reset env: 0.9ms  enqueue env: 0.8ms  process actions: 0.0ms
> 2023-02-08 01:31:59,871 learner: learn: 1605.5ms  synchronize: 1340.3ms  update agent: 255.6ms  compute returns: 7.9ms  after update: 1.6ms  overlap_transfers: 0.9ms
> 2023-02-08 01:31:59,871 policy: rollout: 3256.3ms  wait: 956.7ms  step policy: 82.9ms  batch obs: 2.8ms  update storage: 1.9ms  finish rollout: 0.6ms
> Process ForkServerProcess-35:
> Traceback (most recent call last):
>   File "/home/sha279/fphab/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
>     self.run()
>   File "/home/sha279/fphab/lib/python3.7/multiprocessing/process.py", line 99, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/sha279/scene-builder-datasets/fphab/habitat-lab/habitat-baselines/habitat_baselines/rl/ver/worker_common.py", line 193, in _worker_fn
>     process_class(*process_args, **process_kwargs).run()
>   File "/home/sha279/fphab/lib/python3.7/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
>     return func(*args, **kwargs)
>   File "/home/sha279/scene-builder-datasets/fphab/habitat-lab/habitat-baselines/habitat_baselines/rl/ver/inference_worker.py", line 527, in run
>     self.try_one_step()
>   File "/home/sha279/scene-builder-datasets/fphab/habitat-lab/habitat-baselines/habitat_baselines/rl/ver/inference_worker.py", line 494, in try_one_step
>     t_stamp=t_step_end,
>   File "faster_fifo.pyx", line 143, in faster_fifo.Queue.put
>   File "faster_fifo.pyx", line 138, in faster_fifo.Queue.put_many
> queue.Full
> Traceback (most recent call last):
>   File "habitat-baselines/habitat_baselines/run.py", line 102, in <module>
>     main()
>   File "habitat-baselines/habitat_baselines/run.py", line 54, in main
>     run_exp(**vars(args))
>   File "habitat-baselines/habitat_baselines/run.py", line 98, in run_exp
>     execute_exp(config, run_type)
>   File "habitat-baselines/habitat_baselines/run.py", line 81, in execute_exp
>     trainer.train()
>   File "/home/sha279/fphab/lib/python3.7/contextlib.py", line 74, in inner
>     return func(*args, **kwds)
>   File "/home/sha279/scene-builder-datasets/fphab/habitat-lab/habitat-baselines/habitat_baselines/rl/ver/ver_trainer.py", line 551, in train
>     self.preemption_decider.learner_time(self._learning_time)
>   File "/home/sha279/scene-builder-datasets/fphab/habitat-lab/habitat-baselines/habitat_baselines/rl/ver/preemption_decider.py", line 430, in learner_time
>     (PreemptionDeciderTasks.learner_time, learning_time)
>   File "faster_fifo.pyx", line 143, in faster_fifo.Queue.put
>   File "faster_fifo.pyx", line 138, in faster_fifo.Queue.put_many
> queue.Full

SergioArnaud, 2023-02-10:
I have encountered this issue several times and in general haven't found a way to fully solve it. I've noticed that augmenting the [preemption decider queue](https://github.com/facebookresearch/habitat-lab/blob/7c842d9dec45b4fcd73923c5b2d5c150a8205168/habitat-baselines/habitat_baselines/rl/ver/worker_common.py#L38) size to something like `4 * 1024 * 1024 * num_environments` helps a lot, also reducing the number of environments is helpful. With this two modifications I'm encountering much less this issue.

sanjayharesh, 2023-02-14:
Hi @SergioArnaud ,

Thanks for you reply. I tried the workarounds you suggested but it seems that they only delay this error i.e. the training runs for a bit longer (~12 hrs) before crashing. 

Do we know what causes this? Is there anything else that I can try or keep in mind?

Zeying-Gong, 2024-06-05:
I am also curious about how you deal with this problem. I encountered that when I tried to reproduce the code in hssd.