facebookresearch/habitat-lab #1139: ZeroDivisionError：SPL measure alwary get this problem when using Gibsonv2

GuoPingPan, 2023-02-15:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?

Habitat-Sim: vx.x.x or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

kozhukovv, 2024-02-29:
I Have same problem when vectorize pointnav environment

Yuxin916, 2024-12-04:
I have same problem

xuwenjie401, 2025-03-05:
it's not just Gibsonv2，I'm using pointnav_hm3d_v1，the "DistanceToGoal" Measure can return inf frequently. I checked the code and find that this related to pathfinder 
` def geodesic_distance(
        self,
        position_a: Union[Sequence[float], np.ndarray],
        position_b: Union[
            Sequence[float], Sequence[Sequence[float]], np.ndarray
        ],
        episode: Optional[Episode] = None,
    ) -> float:
        if episode is None or episode._shortest_path_cache is None:
            path = habitat_sim.MultiGoalShortestPath()
            if isinstance(position_b[0], (Sequence, np.ndarray)):
                path.requested_ends = np.array(position_b, dtype=np.float32)
            else:
                path.requested_ends = np.array(
                    [np.array(position_b, dtype=np.float32)]
                )
        else:
            path = episode._shortest_path_cache

        path.requested_start = np.array(position_a, dtype=np.float32)

        self.pathfinder.find_path(path)

        if episode is not None:
            episode._shortest_path_cache = path

        return path.geodesic_distance`
while position_a and position_b are both valid numbers, path.geodesic_distance can be inf（Even position_a & position_b 's y_coordinates are the same, which I assume they are in the same floor）.

So, I am confused, is this because the pathfinder.find_path(path) can't find a navigable way to the goal? but this sometimes occured very frequently, sometimes it won't happen. 

I would greatly appreciate it if you could help
@aclegg3 @erikwijmans