facebookresearch/habitat-lab #1149: How to debug VectorEnv?

GuoPingPan, 2023-02-20:
Every time when there is something wrong with my code, I couldn't find the what had happend because using `VectorEnv` and `multiprocessing` only get the error message as below.
```
Traceback (most recent call last):
  File "generate_dataset.py", line 183, in <module>
    envs.call(func_list)
  File "/home/hello/pgp_eai/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 542, in call
    results.append(read_fn())
  File "/home/hello/pgp_eai/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 108, in __call__
    res = self.read_fn()
  File "/home/hello/pgp_eai/habitat-lab/habitat-lab/habitat/utils/pickle5_multiprocessing.py", line 68, in recv
    buf = self.recv_bytes()
  File "/home/hello/anaconda3/envs/eai/lib/python3.8/multiprocessing/connection.py", line 216, in recv_bytes
    buf = self._recv_bytes(maxlength)
  File "/home/hello/anaconda3/envs/eai/lib/python3.8/multiprocessing/connection.py", line 414, in _recv_bytes
    buf = self._recv(4)
  File "/home/hello/anaconda3/envs/eai/lib/python3.8/multiprocessing/connection.py", line 383, in _recv
    raise EOFError
EOFError
```
and I only found few tutorials about `VectorEnv` like https://github.com/facebookresearch/habitat-lab/blob/main/habitat-baselines/habitat_baselines/common/construct_vector_env.py

so after watch the source code of VectorEnv, I main use it by
```
envs = construct_envs(args)
for i in range(args.num_processes):
    func_list.append('run')
envs.call(func_list)
func_list = []

# run is my own env method
class Dataset_Env(habitat.RLEnv):
    def run(self):
        ...
```
It is right?

Let's rephrase the problem:
1. How to debug the `VectorEnv` that can show what is wrong with my code clearly
just like `ModuleNotFoundError: No module named 'none'`
2. How to use `VectorEnv`? 
just like call the function of self-defined Env and any other function.

aclegg3, 2023-06-02:
Hey @GuoPingPan you should try using a ThreadedVectorEnv which splits workers into threads instead of processes which allows log messages and debug info to be returned and investigated correctly.

Check out this debug flag: https://github.com/facebookresearch/habitat-lab/blob/db1539ac4cb11a1c50cf6bba498f929e8d73daa0/habitat-baselines/habitat_baselines/common/construct_vector_env.py#L96

francescotaioli, 2023-08-09:
Thanks @aclegg3!

**Note**: if your using habitat-lab-challenge, you should manually create the ```ThreadedVectorEnv``` since  setting ```HABITAT_ENV_DEBUG=1``` . To do this, comment the following lines and add the ```ThreadedVectorEnv```:
```
# envs = habitat.VectorEnv(
    #     make_env_fn=make_env_fn,
    #     env_fn_args=tuple(zip(configs, env_classes)),
    #     workers_ignore_signals=workers_ignore_signals,
    # )
envs = habitat.ThreadedVectorEnv(
    make_env_fn=make_env_fn,
    env_fn_args=tuple(zip(configs, env_classes)),
    workers_ignore_signals=workers_ignore_signals,
)
```
in ```habitat-lab-challenge-2022/habitat_baselines/utils/env_utils.py```

GuoPingPan, 2023-08-10:
Thanks a lot for your explanation.

closed 2023-08-10 by GuoPingPan