facebookresearch/habitat-lab #1154: question about the hm3d v1 dataset, goal position is loss

KarelZhang, 2023-02-22:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: challenge 2022 branch

Habitat-Sim: challenge 2022 branch

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I got some questions about the result of top-down map, when I run the eval data of hm3d v1.
The question is that the top-down map does not contain any goal position, is there anything wrong?

the top down map is:
[image]

the episode information is :
episode_id:‘65’
scene_id='<path to hm3d>/val/00891-cvZr5TUy5C5/cvZr5TUy5C5.basis.glb

goals=[
      ....
     object_id = 469,
     object_name = 'plant_469',
     object_category = 'plant',
    ....
]

dhruvbatra, 2023-02-22:
Have you gone through this documentation? 
https://aihabitat.org/docs/habitat-lab/habitat-lab-tdmap-viz.html

KarelZhang, 2023-02-23:
> Have you gone through this documentation? https://aihabitat.org/docs/habitat-lab/habitat-lab-tdmap-viz.html

hi, batra, thanks for your great help.

I have solved problem above by running the /examples/shortest_path_follower_example.py.

The reason that the object goal is not located on the top-down map is it is located on the second floor.

Thanks again !!!

closed 2023-02-23 by KarelZhang