facebookresearch/habitat-lab #1158: Pretrained Weights for Skills

S-Karnik, 2023-02-23:
I would like to reproduce the results of the baselines, but I'm wondering where I can find the pretrained weights for skills. For instance, when I run `python -u habitat-baselines/habitat_baselines/run.py --config-name=rearrange/rl_hierarchical.yaml`, I encounter an error that I have no pretrained weights `data/models/open_cab.pth`. Is there any way to obtain the pretrained weights for skills, other than training for each individual skill? Thank you!

jiminsun, 2023-03-30:
Hi! 

Are there any updates on the pretrained checkpoints of individual skills as defined in 
https://github.com/facebookresearch/habitat-lab/blob/10b898a80240059e551fa2d7f80a244988d7ad95/habitat-baselines/habitat_baselines/config/habitat_baselines/rl/policy/hierarchical_policy/defined_skills/nn_skills.yaml ?

Thank you!

ASzot, 2023-03-30:
Hello, we provide pre-trained skill checkpoints that are compatible with the 2022 Habitat Rearrangement Challenge. See [here](https://github.com/facebookresearch/habitat-challenge/tree/rearrangement-challenge-2022#hierarchical-rl-starter-code) for instructions on how to use them.

closed 2023-03-30 by dhruvbatra