facebookresearch/habitat-lab #1213: Feature request: migrate from gym to gymnasium

elliottower, 2023-03-22:
## 🚀 Feature

Migrate from [gym](https://github.com/openai/gym) (no longer maintained) to [gymnasium](https://github.com/Farama-Foundation/Gymnasium).

## Motivation

[Gymnasium](https://github.com/Farama-Foundation/Gymnasium) a maintained fork of openai gym and is designed as a drop-in replacement (`import gym` -> `import gymnasium as gym`). Beyond just bugfixes, many RL training libraries have also switched ([rllib](https://github.com/ray-project/ray/tree/master/rllib), [tianshou](https://github.com/thu-ml/tianshou), [CleanRL](https://github.com/vwxyzjn/cleanrl)), or are planning to switch ((stable-baselines3](https://github.com/DLR-RM/stable-baselines3/blob/e5deeed16efb57c34ccdcb14692439154d970527/docs/guide/install.rst#bleeding-edge-version)). It would be great if users could train agents on Habitat using the latest models and features from these libraries (e.g., scalable distributed training/model serving using [Ray](https://github.com/ray-project/ray)/[RLlib](https://github.com/ray-project/ray/tree/master/rllib)). 

## Pitch

For information about upgrading and compatibility, see [migration guide](https://gymnasium.farama.org/content/migration-guide/) and [gym compatibility](https://gymnasium.farama.org/content/gym_compatibility/). The main difference is the API has switched to returning `truncated` and `terminated`, rather than `done`, in order to give more information and mitigate edge case issues (for example, many popular tutorials/implementations of Q learning using gym were actually incorrect because of `done`, there will be an upcoming blog post explaining more details about this on the Farama site (https://farama.org/blog).

## Alternatives

An alternative is to keep using gym internally and write a wrapper to be compatible with these other libraries, but migration to gymnasium should not be too difficult and it would be cleaner to avoid the extra dependency. A compatibility wrapper could also be made using [Shimmy](https://github.com/Farama-Foundation/Shimmy), but this is mainly for non-gym projects such as DeepMind Control, DeepMind Lab, etc. 

## Additional context

 
We would love to add this repository to our list of [third party environments](https://gymnasium.farama.org/environments/third_party_environments/) , as it is a major research tool, but are only including repositories which are compatible with gymnasium.

Being able to use habitat environments with gymnasium would also enable a lot of research. It could potentially be integrated into [gymnasium-robotics](https://github.com/Farama-Foundation/Gymnasium-Robotics) as well, which would be helpful for researchers.

aclegg3, 2023-07-17:
We are interested in learning more about the steps involved in this migration and the benefits of doing so. We are also happy to accept contributions and to review PRs adding additional support such as this.

elliottower, 2023-07-17:
> We are interested in learning more about the steps involved in this migration and the benefits of doing so. We are also happy to accept contributions and to review PRs adding additional support such as this.

Great to hear, I could look into it at some point in the next few weeks and see how difficult it looks. It seems you’re using gym v22 which uses a done signal rather than separate termination and truncation signals which are used in v26 and in gymnasium. As for benefits of migration all major training libraries have transitioned to Gymnasium (RLlib, Stable-Baselines3, CleanRL, Tianshou, Google’s Dopamine is planning on it) so it would be desirable for users and researchers to work with your environments. Being unmaintained there are also installation issues on certain systems (M1 and other ARM professors for example, also I know Gym V21 relies on dependencies which are also unmaintained which leads to problems), for Gymnasium we prioritize compatibility with current training libraries and other libraries like numpy/torch, python 3.11, etc. 

We are also finalizing a blog post explaining more about termination and truncation, which explains why the previous way of doing things led to incorrect implementations of algorithms and other unncaught bugs.