facebookresearch/habitat-lab #1218: Semantic segmentation and object detection - mpcat40 (40 categories) to category_mapping (1659)

francesco-taioli, 2023-03-24:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7

Habitat-Sim: v0.1.7 

## Docs and Tutorials

- [x] Did you read the docs?  https://aihabitat.org/docs/habitat-lab/

-  [x] Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

## ❓ Questions and Help
I'm using [Matterport3D](https://niessner.github.io/Matterport/) as dataset with three sensors: 1. RGB 2. Depth 3. Semantic
At the moment, I can retrieve the observation and also visualize the segmented images . If I understand correctly, each semantic instance is mapped into 40 categories defined in [mpcat40.tsv](https://github.com/niessner/Matterport/blob/master/metadata/mpcat40.tsv).

**Is it possible to retrieve the corresponding _label id_ defined [category_mapping.tsv](https://github.com/niessner/Matterport/blob/master/metadata/category_mapping.tsv), to have a more accurate segmentation classification?**

Moreover, is it possible to retrieve the bounding box of the objects mapped in the observation space, like x1 y1 x2 y2 on the RGB image?

whcpumpkin, 2024-02-03:
Hi, 
I also want to obtain a more finer-grained segmentation maps.
Did you find a method?

francescotaioli, 2024-02-03:
@whcpumpkin nope, I didn't find a solution for that.