facebookresearch/habitat-lab #1223: Incorrect ee_target inside ArmEEAction

Murkey8895, 2023-03-28:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.4
Habitat-Sim: v0.2.3

## 🐛 Bug

I'm trying to control the arm with ArmEEAction. Howerver, I find the `ee_target` ArmEEAction which is calculated from `_ik_helper` is not aligned with the end effector. Thus, I can not correctly move the end effector to the desired position.
The visualized `ee_target` and the visualized end effector position using the function `visualize_position` are here:
[image]
[image]

## Expected behavior
The `ee_target` should be aligned with the end-effector

Would you please give any suggestions on this?

Thank you so much!

vantony1, 2024-04-16:
@Murkey8895 Hello, I am trying to achieve similar functionality of the end-effector. could you possibly share your ArmEEAction code?

vantony1, 2024-04-17:
Specifically, I am getting ik_helper = None @Murkey8895