facebookresearch/habitat-lab #1239: Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total

18128381510, 2023-04-04:
when i make a Non-interactive testing: python examples/example.py, it report a issue:
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total
WindowlessContext: Unable to create windowless context
but i have install cuda-11.1 and nvidia-smi  command is working correctly, how can i solve this issue

klammecr, 2023-04-06:
I am also seeing this now. I wonder if they messed with the base container?

HariniNarasimhan, 2023-12-04:
Same issue. Please help

bbjyzzwwy, 2023-12-17:
i had the same problem , I solved the problem by reinstalling the graphics driver, and the NVIDIA-xxx-xxx-xxx.run file should not be executed with the argument --no-opengl-files.