facebookresearch/habitat-lab #1253: Filter semantic annotation in HM3D dataset

jemo-mh, 2023-04-10:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.4

Habitat-Sim: v0.2.4

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hi!

I want to filtering out some classes in HM3D dataset. But edit %Datasetname%.semantic.txt  doesn't work

Dataset
`data/scene_datasets.hm3d/minival/00800-TEEsavR23oF/` there are four files
- TEEsavR23oF.basis.glb
- TEEsavR23oF.basis.navmesh
- TEEsavR23oF.semantic.glb
- TEEsavR23oF.semantic.txt

And I would like to filter just remain ceiling annotation.
When I delete other annotations in TEEsavR23oF.semantic.txt, `habitat_sim.Simulator` generate empty class and printed like 
`[21:50:57:044332]:[Metadata] AttributesManagerBase.h(357)::buildAttrSrcPathsFromJSONAndLoad : No Glob path result for ./data/scene_datasets/hm3d/train/00017-oEPjPNSPmzL/*.basis.scene_instance.json`

And edit annotations to other class in TEEsavR23oF.semantic.txt also doesn't work. 

I think It is related in `*.basis.scene_instance.json` in `hm3d_annotated_basis.scene_dataset_config.json` but I can't find those `.basis.scene_instance.json` file locally. 

Is there any method to filter annotations??
Thanks for your help in advance !

edwardjjj, 2024-03-15:
Hi, Have you resolve the issue? I'm also wondering where to find the scene_instance.json files

rflor01, 2024-06-27:
Has this issue been resolved? I am also modifying my .semantic.txt files and is not making any difference.