facebookresearch/habitat-lab #1398: The model rendered on the Replica dataset is damaged

oskar0812, 2023-06-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?

Habitat-Sim: vx.x.x or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

[image]
I used the habita-sim to render the Replica dataset, but the model in the resulting image appears incomplete.
Does anyone know what the situation is?And how can I correct this problem?

dhruvbatra, 2023-06-02:
Hard to be certain with just one image, but in general, 3D reconstruction is not a solved problem and you should expect missing thin structures.

oskar0812, 2023-06-02:
> Hard to be certain with just one image, but in general, 3D reconstruction is not a solved problem and you should expect missing thin structures.

From various perspectives, structures similar to chairs and chandeliers are all damaged. And shouldn't it be a confirmed model for the Replica dataset? It is not generated through 3D reconstruction.

dhruvbatra, 2023-06-02:
> shouldn't it be a confirmed model for the Replica dataset? It is not generated through 3D reconstruction.

It is. Please read about it here: https://arxiv.org/abs/1906.05797

closed 2023-06-02 by dhruvbatra