facebookresearch/habitat-lab #1438: Interworking the Habitat simulator with the newly captured Matterport 3D environment data

etri-jeon0123, 2023-06-26:
Currently, we want to shoot a laboratory using Metaport Pro3 equipment sold by Metaport Company and operate on Habitat.

However, [Habitat - Matterport 3D Research Dataset] provided by Metaport is driven by Habitat as in image (1),
but the experimental space we photographed with Metaport Pro3 equipment does not operate in Habitat as in image (2).
(The code we tested is Habitat-sim's interactive test (python), https://github.com/facebookresearch/habitat-sim)

Image(1) Habitat-Matterport 3D Research Dataset
[image]

Image(2) Habitat-Our dataset with Metaport Pro3 equipment
[image]

The meaning of "not working" is that the .glb file is executed,
but it is impossible to move forward, backward, left, and right, as shown in image (2).

As far as we're concerned,

Failed to properly generate .nvamesh file provided by Habitat.
Pre- and post-processing of 3D.glb data taken is required.
Therefore, the Habitat developer wants an answer on how to solve the above problem.

aclegg3, 2023-07-17:
Hey @etri-jeon0123,

From image 2 it appears that your resulting .glb asset needs to be rotated by 90 degrees in order to be used in Habitat correctly. Habitat's coordinate system is Y up, -Z forward. 
Because the scene is sideways, the navmesh generated automatically will not be correct.

You can correct this simply by either:
1. adding up/front correcting in the stage config file: https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes
2. importing the asset in a 3d modeling software (e.g. Blender) and appying the rotation manually before exporting. Note that Blender has an option on export to correct the rotation frame.

etri-jeon0123, 2023-07-18:
> Hey @etri-jeon0123,
> 
> From image 2 it appears that your resulting .glb asset needs to be rotated by 90 degrees in order to be used in Habitat correctly. Habitat's coordinate system is Y up, -Z forward. Because the scene is sideways, the navmesh generated automatically will not be correct.
> 
> You can correct this simply by either:
> 
> 1. adding up/front correction in the stage config file: https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes
> 2. importing the asset in a 3d modeling software (e.g. Blender) and appying the rotation manually before exporting. Note that Blender has an option on export to correct the rotation frame.

Thank you for your answer.

closed 2023-07-18 by aclegg3