facebookresearch/habitat-lab #1442: AttributeError: 'dict' object has no attribute 'env_specs' while importing habitat

schowdhury671, 2023-06-29:
Hi,

I am getting the following error while trying to import habitat:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-lab/habitat/__init__.py", line 17, in <module>
    from habitat.core.vector_env import ThreadedVectorEnv, VectorEnv
  File "/fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 37, in <module>
    from habitat.gym.gym_env_episode_count_wrapper import EnvCountEpisodeWrapper
  File "/fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-lab/habitat/gym/__init__.py", line 8, in <module>
    from habitat.gym.gym_definitions import make_gym_from_config
  File "/fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-lab/habitat/gym/gym_definitions.py", line 106, in <module>
    if "Habitat-v0" not in registry.env_specs:
AttributeError: 'dict' object has no attribute 'env_specs'

```
The pip list is as follows:

Package                      Version      Editable project location
---------------------------- ------------ ----------------------------------------------------------------------
absl-py                      1.4.0
antlr4-python3-runtime       4.9.3
astunparse                   1.6.3
attrs                        20.3.0
audioread                    3.0.0
braceexpand                  0.1.7
Brotli                       1.0.9
cachetools                   4.2.4
certifi                      2023.5.7
cffi                         1.15.1
charset-normalizer           3.1.0
clang                        5.0
click                        8.1.3
cloudpickle                  2.2.1
cmake                        3.26.4
colorama                     0.4.6
contourpy                    1.1.0
cycler                       0.11.0
Cython                       0.29.35
decorator                    4.4.2
faster-fifo                  1.4.5
filelock                     3.12.2
flatbuffers                  23.5.26
fonttools                    4.40.0
gast                         0.4.0
gitdb                        4.0.10
GitPython                    3.1.31
google-auth                  2.21.0
google-auth-oauthlib         1.0.0
google-pasta                 0.2.0
grpcio                       1.56.0
gym                          0.26.2
gym-notices                  0.0.8
h5py                         3.1.0
habitat-baselines            0.2.4        /fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-baselines
habitat-lab                  0.2.4        /fs/nexus-projects/ego_data/active_avsep/habitat-lab/habitat-lab
habitat-sim                  0.2.4
hydra-core                   1.3.2
idna                         3.4
ifcfg                        0.23
imageio                      2.31.1
imageio-ffmpeg               0.4.8
importlib-metadata           6.7.0
importlib-resources          5.12.0
jax                          0.4.13
Jinja2                       3.1.2
joblib                       1.3.0
keras                        2.12.0
Keras-Preprocessing          1.1.2
kiwisolver                   1.4.4
libclang                     16.0.0
librosa                      0.9.1
lit                          16.0.6
llvmlite                     0.40.1
lmdb                         1.4.1
magnum                       0.0.0
Markdown                     3.4.3
MarkupSafe                   2.1.3
matplotlib                   3.7.1
ml-dtypes                    0.2.0
moviepy                      1.0.3
mpmath                       1.3.0
msgpack                      1.0.5
munkres                      1.1.4
networkx                     2.2
numba                        0.57.1
numpy                        1.23.5
numpy-quaternion             2022.4.3
nvidia-cublas-cu11           11.10.3.66
nvidia-cuda-cupti-cu11       11.7.101
nvidia-cuda-nvrtc-cu11       11.7.99
nvidia-cuda-runtime-cu11     11.7.99
nvidia-cudnn-cu11            8.5.0.96
nvidia-cufft-cu11            10.9.0.58
nvidia-curand-cu11           10.2.10.91
nvidia-cusolver-cu11         11.4.0.1
nvidia-cusparse-cu11         11.7.4.91
nvidia-nccl-cu11             2.14.3
nvidia-nvtx-cu11             11.7.91
oauthlib                     3.2.2
objectio                     0.2.29
omegaconf                    2.3.0
opencv-python                4.7.0.72
opt-einsum                   3.3.0
packaging                    23.1
Pillow                       9.4.0
pip                          23.1.2
platformdirs                 3.8.0
ply                          3.11
pooch                        1.7.0
proglog                      0.1.10
protobuf                     4.23.3
pyasn1                       0.5.0
pyasn1-modules               0.3.0
pycparser                    2.21
pyglet                       2.0.8
pyparsing                    3.1.0
PyQt5-sip                    12.11.0
PySocks                      1.7.1
python-dateutil              2.8.2
PyWavelets                   1.4.1
PyYAML                       6.0
requests                     2.31.0
requests-oauthlib            1.3.1
resampy                      0.4.2
rsa                          4.9
scikit-image                 0.16.2
scikit-learn                 1.2.2
scipy                        1.11.1
setuptools                   67.8.0
simplejson                   3.19.1
sip                          6.6.2
six                          1.15.0
smmap                        3.0.5
SoundFile                    0.10.3.post1
sympy                        1.12
tensorboard                  2.12.3
tensorboard-data-server      0.7.1
tensorboard-plugin-wit       1.8.1
tensorflow                   2.12.0
tensorflow-estimator         2.12.0
tensorflow-io-gcs-filesystem 0.32.0
termcolor                    1.1.0
threadpoolctl                3.1.0
toml                         0.10.2
torch                        2.0.1
torchsummary                 1.5.1
tornado                      6.3.2
tqdm                         4.56.0
triton                       2.0.0
typer                        0.9.0
typing-extensions            3.7.4.3
unicodedata2                 15.0.0
urllib3                      1.26.16
webdataset                   0.1.40
Werkzeug                     2.3.6
wheel                        0.38.4
wrapt                        1.12.1
zipp                         3.15.0
```

Can someone please provide any help in this regard? Thanks in advance.

Diligent06, 2024-08-15:
I meet the same issue. Did you solve this issue?

Benson722, 2024-12-10:
I meet the problem at the same. But when I try to add:

`import sys
sys.path.insert(0, "")
`

at the top of the code (before the import habitat), it works.

By the way, my environment is docker about "soniaraychau/multion-challenge:starter_2024".

knightzzz9w, 2025-01-03:
Well, it seems just about the gym version problem. your gym0.26.2 (which from the docker image multion-challenge:starter_2024) matches "registry" but not "registry.env_specs", there's update. 

But in the requirement file from the official package: https://github.com/facebookresearch/habitat-lab/blob/a12b42adad66f31932b43f0d14da17cc1013297a/habitat-lab/requirements.txt#L1

daituantuan, 2025-01-03:
Well, it seems just about the gym version problem. your gym0.26.2 (which from the docker image multion-challenge:starter_2024) matches "registry" but not "registry.env_specs", there's update. 

But in the requirement file from the official package: https://github.com/facebookresearch/habitat-lab/blob/a12b42adad66f31932b43f0d14da17cc1013297a/habitat-lab/requirements.txt#L1